mist-devel/mist-binaries #3: ZX Spectrum performance issue

sorgelig, 2015-12-04:
As we talked before, ZX Spectrum core has performance issue which is clearly visible for example on scene demos like Lyra II Megademo (part with dancing girl is slow) and Shock Megademo (first part is slow).
Probably it's because not enough memory access bandwidth of CPU.

harbaum, 2015-12-04:
Can you give numbers? Something like "X percent" more?

sorgelig, 2015-12-04:
If i look at Shock then i would say 100% more. If look at Lyra II then may be 50% more.

sleary78, 2015-12-04:
The best source on spectrum timings is the Harlequin clone project.

http://www.zxdesign.info/harlequin.shtml

Cheers
S

sorgelig, 2015-12-05:
What if core will simulate video memory inside FPGA? There are only two pages with around 8kb each used for video. So, it should be able fit inside FPGA. In this case it's possible to decouple video output from other memory access and thus give more bandwidth for CPU.

sleary78, 2015-12-05:
My advice to solve this issue is to bring everything into verilator with a
simulator similar to the BBC Model i wrote and then see how the fusetests
behave...

http://www.shadowmagic.org.uk/spectrum/fusetest.zip

S.

sleary78, 2015-12-05:
Have a look at this...

I converted all the VHDL code to Verilog in here.

https://www.dropbox.com/s/ifryp98mzsiwhx6/bench.tgz?dl=0

Then I setup a ula test which runs under verilator. It doesnt run the whole
machine yet but it simulates video output for the ULA.

S.

harbaum, 2015-12-05:
Sounds good. I'll definitely have a look at it. Thanks.

harbaum, 2015-12-07:
Runs here. I'll play a little with this. 

In my current version the CPU runs at 3.5Mhz but only gets ram access every second cycle (1.75Mhz) to avoid collisions with video ram access. Imho this should not cause any slowdown as the z80 will never access the bus at full 3.5Mhz

renaudhelias, 2015-12-07:
For implementing VRAM inside FGPA, you can use it as mirror RAM (duplicate
of externel RAM, having just write access from Z80, and read access from
VGA)
That's good against conflicts, and you still can dump full external RAM for
debug purpose...

Regards,

Renaud

harbaum, 2015-12-07:
I like the idea of not using internal ram for the two video areas. But it may be a easy way to tell if giving the cpu full 3.5mhz ram access would actually make a difference ... which i actually doubt

sleary78, 2015-12-07:
Is single byte latency the issue? If so a simple cache should help

harbaum, 2015-12-08:
I actually don't know what the issue is. The original core simply gave the t80 a 3.5mhz clock. I have restricted this in a way that only every second 3.5mhz cycle is available to the cpu to do a memory cycle.

sleary78, 2015-12-08:
That should be fine then. You are correct because basically the z80 only
does something every other clock cycle because of the refresh.

I dont think even the ALU does work during the refresh cycles. So basically
we have double the time to get the data back from RAM as a normal Z80 with
DRAM. The SDRAM latency isnt great but its not *that* bad.

S.

closed 2016-01-09 by sorgelig