mist-devel/mist-binaries #4: Amiga AGA: support for higher resolutions.

sorgelig, 2016-01-09:
I've thought in scandoubler mode i will get progressive mode from "Laced" modes in Workbench, but to my surprise i still see flickering screen. 
Why don't convert interlaced modes into progressive while using scandoubler? 
And what stops Amiga AGA using higher VGA modes since MIST natively support VGA?

sleary78, 2016-01-09:
The scandoubler needs to store a full frame buffer for the deinterlaced mode for this to work. 2 line scandoubling doesn't work for deinterlacing because adjacent lines are a frame apparr

sleary78, 2016-01-09:
Appart even. Damn autocorrect

sorgelig, 2016-01-09:
If i remember correct I was using progressive modes on A1200 with Amiga monitor.
So, at least can enable those hires progressive modes in WB.

harbaum, 2016-06-02:
The progressive modes are present in the 1.2 core and they do work. Just today I submitted a fix for the osd in these "productivity modes".

closed 2016-06-02 by harbaum