mist-devel/mist-binaries #5: MinimigAga Core: OSD off-screen on HighRes non interlaced modes

AmigaIstMiST, 2016-03-19:
Hi everyone, this is my first post on this board. I am not a HW Developer, only a user, but I think it might be helpful to post the issues I have found.

Testet with MinimAGA core 1.2 (affects earlier versions, too)
Using NTSC mode:
1. LowRes (640x200): OSD is centered (OK)
2. HighRes (640x400) interlaced: OSD is centered (OK)
3. HighRes (640x400) non-interlaced: OSD is only partly visible on the rigth side of the display and not usable, it's for example not possible to change disk images.

harbaum, 2016-03-19:
Thanks for reporting this. Now that the amiga core got adjustable video modes the osd needs to be able to adopt to this. Many other cores already have this feature. I'll see if I can port one of these back to the amiga core.

sorgelig, 2016-03-21:
How to enable 640x400 non-interlaced mode? 
I don't see this in WB 3.1.

AmigaIstMiST, 2016-03-21:
You need to pull the "double ntsc" monitor type from the "storage" to the "devs" folder. After this, reboot first.

SpeedD408, 2016-03-21:
I can confirm.  When in DoubleNTSC everything is shifted down and the the
right about 2 inches on my screen.

harbaum, 2016-03-21:
As I said: I know the reason and will fix it asap.

SpeedD408, 2016-03-21:
Great.  Thank you.  I just didn't see that it was confirmed by anyone else,
so I thought I would confirm it.  Sorry for the noise.

harbaum, 2016-03-22:
No problem. I was just trying to say that you don't need to prove that the bug really exists.

harbaum, 2016-06-02:
I just submitted a fix for this for inclusion into the next release.

closed 2016-06-02 by harbaum