mist-devel/mist-binaries #6: Writing to ADF disk Images corrupts them

AmigaIstMiST, 2016-03-19:
I have found this on the game LOOM. At first I thougth, it could be a problem with the game itself, but recently I found a remark of another user on a different forum who complained about this problem, so it might be a general,  problem with the floppy emulation.
I have testet this with several Kickstart versions, turbo modes, cpus and 2 versions of the game.
To reproduce:
1. Start LOOM, press F5, save your game to an empty disk. It works.
2. Exit save screen, press again F5 and save to a new slot. The game freezes, the Amiga must be resartet. After this, the disk has errors. It's not possible to load the last saved game state or to save again. When inserting the floppy image while WB3.1 runs, WB complains about it.

harbaum, 2016-03-19:
The problem is limited to loom? Did you verify that the same problem does not appear in a pc based emulator using the same adf?

AmigaIstMiST, 2016-03-19:
I have only tested this on the MiST so far. The ADF I used to save was not the LOOM ADF, but a new empty disk image. After I had the problem, I formatted the used empty ADF and tried again. The same problem occured again.

harbaum, 2016-03-19:
So _any_ write attempt to _any_ adf image corrupts the image? I must admit that I only used hdd/hdf writing so far ...

rootfather, 2016-03-19:
Seems similar to the issue with the corrupted Workbench Install disk I experienced last year.

AmigaIstMiST, 2016-03-19:
I tried to reproduce the behaviour of the the game LOOM with UAE, but with UAE everythink works fine!
Loading and saving of game states is no problem with UAE. The problem only occurs on MiST/MinimigAGA core.

AmigaIstMiST, 2016-03-19:
I found this interesting report on github:

[https://github.com/wizzmilk/mist-board/issues/32](url)

This user has problems with a corrupt WB Install floppy, too. There is already an interesting comparison between the UAE and MiST floppy drives on this report:

> However, still no clue why it seems that the MiST overwrites some parts of the 
> disk instead of stopping when the disk is full. 
> 
> It also seems that MiST is not that accurate in the emulation of the floppy 
> disk speed. I tested the MiST with the "Ami..Alignment System" test disk and 
> get an average 320RPM instead of the 300RPM the drive should have according to 
> Wikipedia. WinUAE averages at 303-305RPM. Read and write speeds also differ: 
> 
> MiST: 6631 bytes/s write, 19944 bytes/s read
> WinUAE: 4922 bytes/s write, 13900 bytes/s read

rkrajnc, 2016-03-19:
Hi AmigaIstMiST,

I tried running Loom from floppies, and I had no problem making two savegames. The floppy seems OK when used in Workbench (see attached image).
[image]

Did you try using another SD card? It seems that is usually the problem.

That other issue you pointed to was also a result of 'incompatible' SD card, this was a response posted from a user, but latter for some reason deleted:

> Comment #22 on issue 32 by philipp.kemptner: Minimig core crushes Workbench 3.1 Install Disk
https://code.google.com/p/mist-board/issues/detail?id=32

> OK,
 i nailed it down to the SD Cards fault:
The Class 10 Cards i tried don't work (Samsung evo sdhc 32gb, Transcend 8gb sdhc)
A Sony 8gb SDHC Class 4 works (fat32) and a transcend SD(noHC) 2gb works somehow even better (fat16 and fat32) - the system didn't stutter on write as it did with the 8gb card.
So i guess this is not the fault of the core, but the hardware? does anyone know of bigger cards (16 or 32 gb) that work?

AmigaIstMiST, 2016-03-19:
I can see this savegames on my Floppy, too.
Now try to read or overwrite this savegames from _within_ Loom.

I use a SanDisk class 4 SD card. My HD images are on this card, too, but I don't have any problems with them.

rkrajnc, 2016-03-19:
You didn't mention loading before. But still, works for me just fine, loading and overwriting. No freezes either.

Are you sure you are placing the correct disks in the floppy drive? It seems the game wants the blank disk (save disk) in DF0, and then asks for the game disk in DF1, even though I started with the game disk in DF0. Just guessing ...

If you can, try a different SD card.

AmigaIstMiST, 2016-03-19:
Yes, the savegame disk was in df0, I can see the files on it.
I have a different SD card here, I will try it tomorrow and give a report.

Thanks for your responses!

AmigaIstMiST, 2016-03-20:
Ok, I testet with a smaller 2GB Sony SD card. I can reproduce the problem again (floppy turbo off):

1. You have to leave the start screen first (I found this out today, sorry). Go to the village, it only takes you 5 seconds, simply click on the left side of the screen. Load/save there.

2. If the savegame floppy is OFS formatted: I can't save, the screen freezes as mentioned above (you have to save at least twice).

3. If the savegame floppy is FFS formatted: I can save without problems. But when I try to load a saved game, SCUMMVM gives me just an empty error message.

AmigaIstMiST, 2016-03-20:
You mentioned in the other thread you linked above, that only few users have this problem. But I believe only few users actually WRITE to a floppy disk, most users READ from them to play games or install software. Till mentioned above, that he mostly uses hdf images. So do I. I own the MiST for two months now (same SD card), before LOOM insisted to write to a savegame floppy.

AmigaIstMiST, 2016-03-20:
I think we are hunting for an old software bug here.
LOOM runs, when I use SLOW RAM.

I think there is something in MiST that's to fast for LOOM, maybe the floppy, maybe RAM, maybe the combination of them together.

But I think that were the problems of the old days, too.

I close this.

closed 2016-03-20 by AmigaIstMiST