mist-devel/mist-binaries #7: MinimigAGA: It is not possible to enter the characters '\' and '|' using a PC keyboard

AmigaIstMiST, 2016-03-22:
I did not find a way to enter the charcters '\' an '|' on the MiST MinimgAGA core using my PC keyboard.
After googling/qwanting for it, I learned that the Amiga had/has one key more on his keyboard, than the PC has.
I think, it should be possible to enter these characters.    
Maybe it is possible to map this key to F11, too (it is already HELP), or to use the corresponding keys on the PC keyboard.

sorgelig, 2016-03-23:
All my notebooks and separate PC keyboards have key with symbols "\" and "|". And i can enter both symbols in Minimig. Probably you live in region where this key considered as unneeded and removed from keyboards in your region.

sorgelig, 2016-03-23:
http://i.stack.imgur.com/6XXAj.jpg
this key is under backspace key.

AmigaIstMiST, 2016-03-23:
Indeed, you are right!
I have a german keyboard, and I don't have this key on it.

http://bboah.amiga-resistance.info/download_photos/a4000keyb_german_big.jpg
 
The picture above shows a german Amiga keyboard, it has these keys. On a german PC keyboards, these characters are present, too, but they do not have an own key:

https://upload.wikimedia.org/wikipedia/commons/thumb/3/36/KB_Germany.svg/550px-KB_Germany.svg.png

And these keys do not work as expected. So it's a problem for some users only.

AmigaIstMiST, 2016-03-23:
Changed the title to "... german keyboard"

harbaum, 2016-03-23:
The MIST board knows nothing about the symbols on the keyboard. Neither on the USB keyboard nor on the Amiga keyboard. All it knows are keycodes. USB keycode and Amiga keycodes. And it translates between them.

On the Amiga side these are e.g.:
http://wiki.amigaos.net/wiki/File:LibFig34-1.png
and
http://wiki.amigaos.net/wiki/File:LibFig34-2.png

On USB side the codes are defined in this document:
http://www.usb.org/developers/hidpage/HID1_11.pdf

The translation in the MIST firmware happens in the usb2ami table in:
https://github.com/mist-devel/mist-firmware/blob/master/keycodes.h

If you know a better mapping which will give you access to the missing keys, then please make a suggestion. The tricky part is that the mapping has to work with all international and US keyboard layouts as the MIST doesn't know the language of your keyboard nor the language of the AmigaOS you are running.

AmigaIstMiST, 2016-03-23:
What about this:

F11 --> HELP
ALT + F11 --> '\'
ALT + SHIFT + F11 --> '|'

F11 is near the physical location of the original key on the Amiga keyboard.
These symbols are not used that often, I think most people who own a MiST don't know that these characters are missing anyway. But those few people who want to use them have a way to use them.

harbaum, 2016-03-24:
As i said: The MIST board only knows about keycodes. A translation of the kind you suggest is not directly possible. Once you press the "Alt" key the keycode of that key is forwarded to the Amiga core. So once F11 is being pressed by the user the Alt (and/or shift) key has already been reported to the Amiga. 

And finally: If we remap single combos like you suggest, what would happen on other international keyboards? You are talking about a german keyboard. Another keyboard may have completely different labels on the keys you are talking about.

I really think we need a way to remap entire keys. There are unused keys above the PCs cursor block. Imho it's better to use one of these.

fordp2002, 2016-03-24:
Why not allow the key mappings to be optionally loaded from SD Card ?

harbaum, 2016-03-24:
US users never had this problem since the US PC keyboards have a backslash key which international keyboard lack. 

The PCs insert key above the cursor block was unused in the amiga core. I mapped it to the backslash key now. So the backslash now isn't left but right of the backspace key. The new version is in the github repository.

sorgelig, 2016-03-24:
> US users never had this problem since the US PC keyboards have a backslash key which international keyboard lack.

i would say, international keyboards have this key. My keyboards aren't US. They are Russian and Taiwanese. All of them (even tiny one) have this key.
So, the problem exists only in some specific regions. IMHO shouldn't be hard to find keyboard in other country to fix this problem. Especially in EU, just order online the keyboard from other EU country and you are set.

> There are unused keys above the PCs cursor block. Imho it's better to use one of these.

May be as optional way only. Please don't require these keys, because there are many RF reduced keyboards lacking extended keys such as calc part and above-cursor keys.

rkrajnc, 2016-03-24:
Lacking a better approximation, the minimig keyboard should match WinUAE keyboard mapping as much as possible.

There's no need to change the default mappings, since there's support for keyboard remap in the ini file:

https://github.com/mist-devel/mist-board/wiki/DocIni

Keep in mind that the minimig mapping is also dependent on the selected keyboard in AmigaOS.

harbaum, 2016-03-24:
The recent patch didn't disable anything. It just mapped a key that was so far unused in the Amiga core onto one used by the core. Unless you hit the insert key there'll be no difference at all for you.

But i agree that the WinUAE mapping may be cool. Which key on an international keyboard is mapped to the amigas backslash key?

harbaum, 2016-03-24:
Sorgelig: Where is the key physically located that maps to backslash on your keyboard? And which USB keycode does that key emit?

On a USB US keyboard the backslash key emits keycode 0x31. On an international keyboard there is no key emitting this code (and indeed, AmigaIstMiST tried all keys on his german keyboard and none emitted this code).

harbaum, 2016-03-24:
It seems russian and taiwanese keyboards are in fact US keyboards with different labels. Some USB specs differ between US and European keyboards and some between US and international keyboards. Slightly confusing with respect to keyboards that neither come from the US nor from Europe ...

sorgelig, 2016-03-24:
The code from this key is 0x5d according to MiST UART console.

sorgelig, 2016-03-24:
> Where is the key physically located that maps to backslash on your keyboard?

this key is always under backspace (between backspace and enter).

AmigaIstMiST, 2016-03-24:
Mapping to an unused key is of course the best solution :-)
I didn't know that such a key existed.

I just uploaded the new version to my MiST, and it works!

The key location really "feels" good, the key is still near the backspace key, but on the right side.
It will be much easier for people to write C programs on the Amiga core, now.

Wow, that fix really came fast.

I close this ticket.

AmigaIstMiST, 2017-01-13:
Issue open again in firmware 170113.

sorgelig, 2017-01-13:
what's the latest FW where it worked?
Actually, easier fix is to buy a US keyboard. Most cores written for US/UK layouts and you will always have problems with German keyboards. It's a question of couple bucks.

sorgelig, 2017-01-13:
New FW supports per-core settings in INI, so you can try to remap some other key for Minimig core only.

AmigaIstMiST, 2017-01-14:
The firmware version I had before was 160910.
Sorry, buying an UK keyboard is not a solution for the Amiga core (and not for the Atari ST core either). Unlike the C64, the Amigas (and Atari STs) sold in Germany had a german keyboard layout with all required special german characters on it.
Till already solved the problem with his fix, everything was working fine.
I noticed, that the behaviour of the F12 key was changed, too, it is not wrongly recognized as an Amiga key any more. Maybe it was broken with this fix?

sorgelig, 2017-01-14:
At first, MiST is not German keyboard centric. If some localized keyboard doesn't work it doesn't mean everything is ruined. Main keyboard even for Amiga is not German, believe me. And German keyboard is not the feature converting Amiga from toy to computer.
Second: i don't remember i changed anything for normal PC keyboard. I've added better support for A1200 native keyboard, but PC keyboard should remain as is.
Actually, there were other changes since 160910 made by other people.

I've just re-checked F12 with my wireless PC keyboard on latest FW and Minimig core - it opens OSD menu as it supposed to be. So what's wrong here?

Actually, before you closed the issue there were no info how the issue was fixed, so i even have no idea how it worked before.

sorgelig, 2017-01-14:
Was that "Insert" key mapped to \ key?
Yes, i've changed it back to insert which is mapped to help key on A1200 native keyboard (Keyrah).

Will see what can i do here..

AmigaIstMiST, 2017-01-14:
I don't know, if a "main" keyboard layout exists. It is nearly impossible to buy a US keyboard in a store here in Germany, and I can imagine it's the same in France and Italy. Maybe it's different in Sweden or in The Netherlands. So this is really a serious bug for many people, if you want to do something in C or BlitzBasic2.

Until recently, F12 wrongly triggered the PAL/NTSC screen mode on the Amiga Boot menu, this is fixed now. I did a bug report for this on chaos repository, and closed the bug yesterday, when I tested the new minimig BETA core. Maybe this was fixed by accident by the keyrah changes?
 
(read your answer, thanks :-) )

sorgelig, 2017-01-14:
Fixed in today's release.

AmigaIstMiST, 2017-01-14:
Just tested it, key works again.

closed 2017-01-14 by AmigaIstMiST

jarcane, 2018-03-13:
I'm using a Finnish keyboard and having this same problem. Neither core currently posted to the repo will allow me to type a backslash, which rather makes C programming difficult ...

harbaum, 2018-03-13:
This is likely not core related but a problem with key mapping in the firmware. 

There are some wiki pages in the mist repository which explain how to remap keys. This might help you.

jarcane, 2018-03-14:
I was able to fix it by installing a Finnish keymap from Aminet (the official Swedish one doesn't have a # symbol). Seems to work so far but I only had a few minutes to mess with it. If I don't like the layout, I've got KeyMapEd now so we'll see how that works.