mist-devel/mist-binaries #9: MinimigAGA: strange dots (graphic issue)

AmigaIstMiST, 2016-03-24:
[image]
[image]
[image]

AmigaIstMiST, 2016-03-24:
I noticed that there are strange dots on GURUs and on the Kickstart 2.0 start screen.
I sadly do not own a real Amiga any more (I sold it and bought a PC for it, I admit it), but I can't remember having seen these dots. When the red guru rectangle is visible, there is also a surplus dot on the left lower side. When the rectangle is not visible, there is a single red dot visible.

AmigaIstMiST, 2016-03-26:
Tested this with Kickstart 2.0/3.1 ECS/OCS.
Kickstart 1.x doesn't have this problem. I didnt't verify this with UAE, I will try to do it in a few days and will report the results here.

AmigaIstMiST, 2016-04-01:
On UAE Kickstart 2.0 startscreen looks fine, no white dot visible (I didn't check gurus).

AmigaIstMiST, 2016-04-10:
https://github.com/rkrajnc/minimig-mist/issues/38

renaudhelias, 2017-06-29:
If the dot disappears after unplug/stop plateform during one day long, it's then certainly due to a memory zone not initialized correctly.
The dot position seems being at begin of VRAM (it's to say at begin of drawable image on screen), isn't it ?

AmigaIstMiST, 2017-07-01:
Hmm, have a look at the 3rd picture, there is a surplus dot _under_ the guru, not on the first position of the drawable area. The behavior is also different between AGA and ECS, like the shifted screen bug.

renaudhelias, 2017-07-01:
Some display could start at another offset like in CPC, but begin of RAM (used for video, alias VRAM) is still begin of RAM. The better to do for the author here is to isolate this pixel forcing it on or else off during RAM (VRAM) read to find firstly that there is only one address of the VRAM that is concerned here just before display. This way he could, in a first step, capture this bug.

renaudhelias, 2017-07-01:
It feels perhaps more a "pixel displayed after the last one" defect.

AmigaIstMiST, 2017-07-21:
To reproduce the bug:
- use ECS or OCS chipset
- use kickstart 3.1 for Amiga 500
- guru somehow :-)
The dot appears in VGA and in SCART configuration.

The dot does *not* appear, if I use kickstart 1.x (strange, isn't it?) and ECS/OCS.
The dot also does not appear, if i use AGA chipset.

renaudhelias, 2017-07-21:
a beta version is a version no finalized yet (but that does something cool), I think it's the way to go.

renaudhelias, 2017-07-21:
https://github.com/rkrajnc/minimig-mist/issues/44

AmigaIstMiST, 2017-07-21:
The dot also appears in 1.2, it was not introduced in the beta version.

AmigaIstMiST, 2018-05-30:
Fixed in V 1.2.3

closed 2018-05-30 by AmigaIstMiST