mist-devel/mist-binaries #10: MinimigAGA core (wording): 68010 CPU, MHz missing

AmigaIstMiST, 2016-03-26:
- Info text on OSD says: "... Minimig can emulate an 68000 or 68020 processor .."
The 68010 cpu is missing here.

- I personally think, there should be MHz after the cpu names. What, if you combine a 68020 cpu with OCS/ECS? Will it be clocked at 7 MHz or 14 MHz? If the text would say "CPU : 68020 (14MHz)" everything would be clear.

(BTW: a 68020 CPU with more MHz (25? 28?) would be cool :-) )

sleary78, 2016-03-26:
There is very little point in emulating a 68010. The only difference from a
68000 is the exception stack frame AFAIK.

AmigaIstMiST, 2016-03-26:
The 68010 core is already there :-)
Only the info text is too old.

AmigaIstMiST, 2016-03-26:
This issue is only about wording.

closed 2021-07-16 by squidrpi