mist-devel/mist-binaries #11: [Amstrad core] scanlines

bonnefoi, 2016-03-28:
Hello,

could you add support for the scanlines and for memory's sake a monochrome option ?

best regards,
Pierre-François.

renaudhelias, 2016-03-29:
Hi,

Original scanlines CRTC option is not implemented yet (Wolfenstrad demo), but is in TODO-list.
Monochrome option added to TODO-list (TODO-list is here : http://www.cpcwiki.eu/index.php/FPGAmstrad#Effort_done)

Regards,
Renaud.

bonnefoi, 2016-03-31:
Perfect :)

renaudhelias, 2016-12-31:
r005.8.9 : green.

renaudhelias, 2017-01-05:
[image]

renaudhelias, 2017-01-05:
r00.5.9.2 "Les Sucres en Morceaux" - green screen

bonnefoi, 2017-01-06:
Thank you very much and happy new year !

renaudhelias, 2017-01-27:
About triangular pixels filter effect, it cannot be done in CoreAmstrad : Amstrad true resolution is 768x288, not 384x288. Then my 768x288 is displayed as 800x600 VGA (one pixel is one pixel), I cannot insert more pixel to reach an great effect.

... but ... something like that could be done : http://www.brunofreitas.com/node/36
[image]
[image]
One line out of two is black, or darker. Let's just have to know "how much darker".
sampling 2 pixels on this picture : (78,153,211) [darker than] (105,190,249)
105/78=1.34
190/153=1.24
249/211=1.18
(1.34+1.24+1.18)/3=1.25=5/4
4/5=0.8=0.5+0.25 (sum of two shift right : 1/2 + 1/4 or else simply another array of input/ouput)

renaudhelias, 2017-01-29:
**Done** (finally just 1/2 darker in second lines gives good results).

Can be closed.

renaudhelias, 2017-01-29:
[image]
[image]

sebdel, 2017-01-30:
That's great, looks exactly like my actual CPC464 :)
If you have some bits left in the extended status register you could let people select the strength of the effect. It's very well done in the spectrum core.

closed 2017-05-13 by renaudhelias