mist-devel/mist-binaries #12: MinimigAGA 1.2: Startup Screen not visible on some monitors (PAL/NTSC)

AmigaIstMiST, 2016-04-01:
It looks like the startup screen of the MinimgAGA core (boing ball with minimig developer names) is using PAL/50 Hz, so some users have problems seeing it (as PAL/50 Hz does not give a valid VGA signal).
I think it would be better, if the startup screen would use NTSC/60Hz, so it would be visible on _any_ monitor.
(Some users, like me, are more or less forced to use NTSC, the behaviour of my monitor is a bit complicated, I can use PAL, but it requires some work every time I switch on the MiST. As 80% of the games also work on NTSC, I am a happy MiST user nevertheless :-) ).

sorgelig, 2016-04-02:
Minimig doesn't follow screendoubler_disable flag on startup screen, so it always uses doubler with 30KHz. 
I also would like to get it fixed, but so far no attention given to this long time already.

efeist, 2016-04-19:
This is an issue for me now that I've switched to an RGB monitor.

AmigaIstMiST, 2016-08-13:
Sorgelig already reported this in chaos repository, duplicaze.

closed 2016-08-13 by AmigaIstMiST