mist-devel/mist-binaries #14: OneChipMSX core still has SD card writing issues

fysticle, 2016-04-12:
The last change to the Mist version of this core states the author fixed the SD writing bug. This doesn't seem to be true for me. Anytime there is an attempt from the core to write to the SD card, the computer locks up. From MSX-DOS a simple "mkdir tmp" will trigger it.

harbaum, 2016-04-13:
Do other cores writing to the SD card work for you?

efeist, 2016-04-13:
I leave this SD card dedicated to the MSX core since it requires FAT16. I'm going to try another card this afternoon and see if it makes a difference.

efeist, 2016-04-13:
Yeah, prepped a brand new SanDisk card with a 2GB Fat16 partition, tried to mkdir foo and it locked up.

sleary78, 2016-04-13:
be really useful to get a debug output from the ARM while this is
happening. Is there an rs232 port on the production MiSTs?

S.

harbaum, 2016-04-13:
I mean if _other_ cores can write to sd card

efeist, 2016-04-13:
Well yes, the Amiga core writes fine to the FAT32 SD card it's on.

harbaum, 2016-04-14:
Ok, then it's really the core having a problem and not the entire SD card signal chain itself. I'll have a look ...

robinsonb5, 2016-04-15:
I haven't been able to reproduce this so far - writing seems to work OK here, but I've only tested with one card so far.  (The upstream comment about fixing SD Writes shouldn't have applied to the MIST, by the way - the issue it fixed only affected the Chameleon port.)

One thing to check - the cards you're using are definitely SD and not SDHC, yes?

efeist, 2016-04-17:
I am embarrassed. All my SD devices are actually SDHC. I've got an actual 2GB SanDisk SD card coming from Amazon and we'll see if this solves my issues, and I assume it will. Oops..

robinsonb5, 2016-04-17:
There's a good chance that's the problem - the OCMSX core can't handle SDHC at all (or at least, the OCMSX's BIOS can't - the BIOS itself can be loaded from SDHC just fine), and the MIST is the only platform on which you can use an SDHC even for reading, since it does translation in the firmware.  There's a warning message when you attempt to write to the SDHC in SD mode, but because the OCMSX core doesn't (yet) suppor the MIST's OSD you can't see it.  Adding support for the native OSD is on my To Do list.

harbaum, 2016-04-17:
That's actually the reason why I don't expect the sdhc card to be the problem. But I'll take a look as well.

robinsonb5, 2016-04-17:
I thought the SDHC -> SD translation was designed to block any attempt to write?  Is that not the case?  (The SDHC emuation must be active, because the SD BIOS on OCMSX doesn't support SHDC at all.)

harbaum, 2016-04-17:
Ah,  indeed, we are talking about writing. Yes,  then a non-sdhc card should solve the problem. :-)

efeist, 2016-04-19:
That was it! Regular SD card is working perfect!

closed 2016-04-19 by efeist