mist-devel/mist-binaries #15: plus_too core ver 151119 fails correctly write cfg file

EricGus, 2016-04-14:
The plus_too core has a bug where it fails to save a proper cfg file.. you end up with a cfg file that is 1 byte in size.  (see attached zip file) {this is with MiST firmware_160324}
[PLUS_TOO.CFG.zip](https://github.com/mist-devel/mist-binaries/files/218569/PLUS_TOO.CFG.zip)

harbaum, 2016-04-14:
How many bytes do you expect to be written?

EricGus, 2016-04-14:
I would suspect more than one. The settings are not retained so I am guessing there is a bug of some kind. If you can try it yourself with a new plus-too Setup. 

Other cores have CFG files that are bigger. Also if you examine the attached CFG file you will see it pretty much void of any info. I.e. No dsk or img file names etc.

harbaum, 2016-04-14:
All the simple cores only write a 1 byte config which simply represents the a bit per OSD line.

The cores with bigger cfg are atari st and amiga. I byte is good.

EricGus, 2016-04-14:
So the cfg file for Plus_too doesn’t retain the disk1 / IMG setting? .. so you have to specify those every time you start the core?

What is happening is if you set those .. and save the config, when you restart MiST/plus_too the disk and img settings you selected are not retained.

harbaum, 2016-06-02:
Yes, the disk settings are not saved. If it would be easy I'd just add it. But it needs more work as the mist isn't a full blown pc and the firmware curently lacks some of the file handling capabilities required for this.

closed 2021-07-16 by squidrpi