mist-devel/mist-binaries #20: AtariST: unstable picture when using SCART cable

AmigaIstMiST, 2016-05-18:
I am using the latest AtariST and system binaries.

When using the AtariST core with a Minimig SCART cable, every 4-5 seconds the image becomes unstable. It is a small flickering, it looks like the image wants to "roll" from top to bottom and is then catched again. It's not a big problem, but clearly visible.
I do not have this problem when using the Amiga or C64 core with a SCART cable.
I did not have this problem, when I used the AtariST core with a VGA cable.

It looks like the video signal of the AtariST core is somehow different from the other cores.

harbaum, 2016-05-19:
All the cores have a different timing as they try to reproduce the timing of the original machines which already differed significantly.

It'd be interesting to know if your screen has the same problem with the video signal of a real Atari ST.

AmigaIstMiST, 2016-05-19:
No sorry, I don't have a real AtariST to test against.

I tried the Atari 800 core for the first time yesterday (new PAL version from the provided link, not from the MiST repository) , it has the same issue.

I don't want to go so far to say, that Atari graphic is generally inferior to Commodore graphic, but on the MiST: yes ;-)

EricGus, 2016-05-19:
For what it's worth the current Commodore 64 core on NTSC mode doesn't work either. (The old core worked but the most recent core just throws a black display when you set it to NTSC)

harbaum, 2016-05-20:
Yeah. The issue was that the video signal of the old c64 core wasn't stable and it had a timing jitter. Some screens wouldn't care for that but many were confused. I had to fix that. But of course that affected the overall timing. And in some cases this will actually break a working setup.

The changes between those versions were limited to the video timing generation. So if the old version worked for your there's no reason to try to use the new one.

EricGus, 2016-05-20:
Understandable but what if new features are added to the core? will you revert back to the old video generation code? I had several people test the NTSC option and none of them worked and folks were using a variety of different output displays. (PAL was fine.. NTSC black screen in all cases)

harbaum, 2016-05-20:
I'll have a look at the c64's NTSC output. Maybe I totally broke it .... does NTSC only fail with the scart cable?

EricGus, 2016-05-20:
No SCART cables here in the USA Till.. can't try it.

EricGus, 2016-05-20:
For what its worth PAL mode seems to work ok.. it just seems NTSC is totally broken..

EricGus, 2016-05-26:
Til for what its worth Sorgelig just recent made some major improvements to the fpga64 core you might wan to check the Atari forums .. though he can't "see" the NTSC issue himself (I asked him about this issue since he was working on the core) .. but you might want to get a copy of his source before publishing any new 64 cores. just an FYI..

harbaum, 2016-05-26:
Thanks for that hint. I just asked him for the code.

EricGus, 2016-05-26:
Sorgelig is doing some amazing things with MiST .. he's incredible .. the most recent "Sorgelig" c64 core is fantastic (save for the NTSC issue)

harbaum, 2016-05-26:
Yes, indeed, he is.

I just asked Peter Wendrich to put the original FPGA64 code under GPL. This would significantly easy further development of the c64 core.

AmigaIstMiST, 2016-08-15:
What about MEGA65 (http://mega65.org)? They are building an enhanced C65 right now. Maybe it's possible to port the C64 only part to the MiST? It's under an open source license (LGPL), and it looks like the project has some momentum.

sorgelig, 2016-08-16:
I couldn't find info about MEGA65 board. Since it's claimed to be LGPL, they should provide circuit design.. Site i weird: a lot of "wow, we are cool." And no info about board itself...

sorgelig, 2016-08-16:
Ok, it seems they use Nexys4DDR dev board.
FPGA used in that board has a lot of FPGA RAM (~5MBit) which is almost 10 times more than in MiST's FPGA. I didn't explore Mega65 source, but according to specs, whole C64/C65 RAM can be put into FPGA RAM which is impossible on MiST board.

AmigaIstMiST, 2016-08-16:
What I meant was: the Mega65 has a builtin C64 mode. Maybe it is possible to make a LGPL C64 out of this part? I did not suggest to port the whole Mega65 to the MiST.

AmigaIstMiST, 2016-08-18:
I noticed, that the issue started with an Atari display issue, and is about C64 now.

As we are completely off topic anyway: I really would like to know, how big the MiST community is. Are there any known numbers, how many MiSTs were built at Lotharek up to today?

AmigaIstMiST, 2016-08-26:
O.K., It looks like most people are more interested in C64 than in AtariST ;-)
Seriously: can anything be done about the unstable AtariST picture? I don't know how whidespread this problem is, maybe only few people have this problem, and I am one of them. I don't know the difference between the Atari signals and the Commodore signals, maybe they are weaker? If yes, maybe the signal can be boosted? Does anybody except me use AtariST+SCART and can tell his/her experience?

AmigaIstMiST, 2017-01-02:
Problem solved by using an external SCART to HDMI converter. So it was a problem with the SCART input of my TV. Sorry for all the noise.

closed 2017-01-02 by AmigaIstMiST