mist-devel/mist-binaries #21: Fpga64: Blue Screen Of Death

AmigaIstMiST, 2016-05-20:
Steps to reproduce (latest fpga64 core):

1. Start fpga64
2. Load and run the program "Simons Basic 3.1" from a D64 image
3. After Simons Basic start screen is visible: press the RIGHT button on the MiST for a few seconds and release it (reset). Result: no reset, but C64 freezes.
4. Switch the MiST of , wait a few minutes, switch it on again, load FPGA64 core. All you see is a blue screen (see screenshot).

From now on it's impossible to use the C64 :-)
You can load other cores (Amiga, AtatiST) and use them. But after you load the fpga64 core the BSOD is back.

Good news: after 1 day, you can use the C64 core again (well, I don't know the minimum time, maybe it's only 1/2
[image]
 hour, but it's more than a few minutes!)

AmigaIstMiST, 2016-05-20:
Good news: OSD works (see screenshot), so you can load other cores. Only the C64 itself is unusable.

EricGus, 2016-05-20:
Did you try loading an older version of the C64 core as well?

AmigaIstMiST, 2016-05-20:
No, I only tried the newest core. It tried a few minutes ago: after 15 minutes the C64 was usable again (rest electricity in RAM?). It would be interesting if this bug is only related to Simons Basic, or if it's a more general problem.

harbaum, 2016-05-20:
Several cores have similar effects. The reason is that sdram is actually loosing it's contents much slower than one would expect. So if a computer checks for some magic values after powerful it may still find them. And if they point to some problematic code they will reproducible lock up until the memory looses its content. 

One solution is to boot a different core that draws this particular part of ram. The best core for this is the Amiga with 24 my fast ram enabled. This will erase most of the 32mb in the mist.

I could actually write a ram eraser core for this purpose ...

AmigaIstMiST, 2016-05-20:
I already tried the Amiga AGA core with 24 MB :-)
It had no effect, the C64 stayed unusable.

harbaum, 2016-05-20:
Then the Amiga doesn't erase it's ram properly. Bad Amiga ...

AmigaIstMiST, 2016-05-20:
Well, does anybody know of a AtariST with 24 MB of RAM? ;-)

Maybe the reset function is the problem? I don't know, what it does, does it trigger a builtin C64 function that is moved to a different location by Simons Basic? AFAIK Simons Basic "enhances" the builtin Basic.

AmigaIstMiST, 2016-05-20:
If it's like this, maybe it wouldn't be a bug in the core, because an original machine would probably behave the same way.

AmigaIstMiST, 2016-05-20:
Maybe it would be a solution, that the whole RAM is erased, when the machine is switched on, just before a core is loaded?

harbaum, 2016-05-20:
Only the fpga can access the ram. So only a core can erase it. Either the target core itself or some 'eraser core'. We are dealing with real hardware here.  It not like some simple emulator where the main cpu can simply erase the ram before it loads the emulator :-)

harbaum, 2016-05-20:
Currently the Atari st is limited to 14. I planned to make an extended ST with 30mb ram. But whenever I start working on that things get more complicated than expected .... maybe I should try again ...

AmigaIstMiST, 2016-05-21:
What about the new "menu core" (haven't tried it yet)? If it's a common problem among different cores, this migth be the right place for a RAM erase function.

harbaum, 2016-05-21:
I wouldn't call it a common problem.  I myself have seen it two or three times.  Buy yes, it exists and the menu core actually sounds like a good place for this. I'll have a look.

robinsonb5, 2016-05-21:
The One Chip MSX core might benefit from this, too - it can occasionally get into a state where it doesn't boot the BIOS, instead showing a black screen and three clicks through the audio.  Power cycling usually fixes it, but being able to do it from a menu core would be ideal.

AmigaIstMiST, 2016-05-22:
My experience with the FPG64 is, that it is impossible to load some games after some time. Normally '0' is the NOP instruction, I wonder how many assembly written software for the C64 exists, that relies on a zeroed RAM? 
I experienced the following behavior with California Games and Lemmings: sometimes the screen stays black, sometimes you see the intro and then the C64 stops, sometimes you can play the the game.

AmigaIstMiST, 2016-09-09:
It looks like the latest changes from sorgelig to the reset function fixed this issue. I testet it with Simons Basic and Zaxxon, the C64 stays usable after a reset now. Nice work :-)

So I close this issue

closed 2016-09-09 by AmigaIstMiST