mist-devel/mist-binaries #22: speedloaders support & 1541 compatibility

mambrino, 2016-05-20:
Hi,

I'm trying to change internal roms to JiffyDOS ones (as indicated in original FPGA64_027 with C1541_SD project guide), but after patch original sources and recompile the whole project always have disk access errors and cpu freezes when loading apps or games. Only with C64 Kernal rom (and original 1541 rom) all seems to work perfectly (no speedup, obviously). Maybe core patch for MiST is not working properly or 1541 emulation needs to be checked? I've test other speedloaders (C64 side only) and also have compatibility issues at disk access process.

Thanks ;)

NML32, 2016-05-20:
Good luck with getting JiffyDOS working as I too would like to see an option for JiffyDOS in the menu and the roms stored on the SD card. (If possible)

harbaum, 2016-05-21:
Yes, i have experienced the same problem during development. These speeders make use of the fact that the speed of the c64 cpu and the 1541 cpu are well known and they replace the communication handshake between both by some simple scheme with a fixed timing. This relies on the c64 and the 1541 cpus to run on exactly the right speed. My assumption is that the speed ratio between both isn't correct in our setup. But I never figured out what  exactly is broken.

mambrino, 2016-05-26:
Recently Sorgelig has done a great work improving 1541 compatibility and adding support for external JiffyDOS rom to this core: http://atari-forum.com/viewtopic.php?f=101&t=29788&start=50

Now works perfectly with speedloaders and this issue can be closed.

Thanks ;)

closed 2016-05-26 by mambrino