mist-devel/mist-binaries #23: OneChipMSX: MSX2 config & compatibility

mambrino, 2016-05-20:
Would be possible to compile a version of MSX core with slot/subslots and rom config like standard MSX2. It will be more compatible with certain software and games than actual MSX2+ config.

Thanks ;)

robinsonb5, 2016-05-20:
You will need to be more specific about what that actually means.  How should the slots be set up?
What about the ROM config needs changing.

(I'm an FPGA developer, not an MSX expert - this request might actually be better directed at an MSX specialist - maybe KdL at msx.org?)

efeist, 2016-05-20:
I think the poster means how in the MiniMig core you can "insert" ADFs from the OSD to the floppy drives vs with the 1Chip core you have to use the ep software emulator. This makes games like Snatcher SD essentially unplayable to my knowledge.

mambrino, 2016-05-21:
Many of MSX1/2 disk games i've tested with basic loaders don't work in MSX2+ (tested in emulator too, maybe is problem of EP disk emulator, but many of these games have a simple loader and only 1 disk):

-Auf Wiedersehen Monty
-Beam Rider
-Boulder dash
-Congo Bongo
-Decathlon
-Desolator
-Donkey Kong
-Hunchback
-Keystone kapers
-Krakout
-Rastan Saga
-Retaliot
-River raid
-Robot wars
-Rolling Thunder
-R-Type
-Sorcery
-Temptations
-Time Curb
-Xevious
-Zanac EX

This can be an interesting slot config, taken from MSX2 emulator config & tested to be compatible with many software:

[image]
[image]
[image]