mist-devel/mist-binaries #24: MinimigCore: Load Configuration doesn't work anymore

sblendorio, 2016-06-24:
With this firmware, while using minimig, and I change configuration through "Load Configuration" option, the kickstart is loaded but:

- Hard Disks are not detected
- F12 (and other ways to show OSD menu) does not work anymore

sblendorio, 2016-06-24:
*updated, specified firmware_160619

Newsdee, 2016-06-24:
Hi,

I'll have a look when I can. Does it work if you revert to the previous
firmware?

sblendorio, 2016-06-24:
Yes, reverting to the previous firmware it all works as before.

Newsdee, 2016-06-24:
OK, I can replicate the problem. The last good firmware is 160610.

I'll go through changes in 160613 to see what caused this. It looks like
even saving a new config and loading it back does not work; although the
original (startup) config is still working fine.

Yes, reverting to the previous firmware it all works as before.

sorgelig, 2016-06-26:
@Newsdee 
did you fix this problem? I'm holding ZX core release because it requires FW update, so i think, it would be good to wait this bug get fixed.

Newsdee, 2016-06-26:
I'm working on it now. I'm not sure if it is my code or a bug in the core,
but I'm reworking all my latest changes to avoid the problem.

Will release a core disabling some features and fixing the Amiga load
config sometime today.

Newsdee, 2016-06-26:
Load Config is now fixed.

Turbo function for button has been disabled for now - will add it back
later on.

closed 2016-06-27 by Newsdee

sblendorio, 2016-06-29:
Thank you!
P.S. I published a review about MiST, hope you'll like it!
http://www.retroacademy.it/2016/06/28/news-eng/mist-y-amiga/2649/