mist-devel/mist-binaries #29: [Minimig-AGA core] Stripes pattern in background colors

mambrino, 2016-07-25:
Hi,

I found an annoying problem. With scandoubler activated in dark consistent color backgrounds a kind of bar/stripes pattern appears. I've tried this with Minimig board and doesn't occur.

Here is a picture in which you can see the effect:

[image]

And this is with minimig board:

[image]

Tested with different cables / monitors, and it's not a problem related to this.

Thanks ;)

rkrajnc, 2016-07-25:
Which core version are you using? 
If this is not a new issue, it should be fixed in the latest release,. Do you have dither enabled?

mambrino, 2016-07-26:
Minimig-AGA v.1.2
I've tested with & without dither / loRes-hiRes filters and problem persists.

Thanks ;)

closed 2021-07-16 by squidrpi