mist-devel/mist-binaries #30: 15Khz for all cores by default?

lroby74, 2016-09-06:
Hello everybody,
just guessing why it's not possible to have 15Khz compatibility on default for all cores, in this way people may use any cores with own RGB Scart monitors like Philips CM8833 or Commodore 1084S or similar.

Thank you all
Roberto

PS
Mist is awesome!

harbaum, 2016-09-09:
The cores that do support 15khz all support the same mist.ini setting. So this single setting affects all cores that support it.

But not all cores support 15khz if that's your question. This has to be implemented in each and every core.

lroby74, 2016-09-09:
Infact i mean this, why 15 khz feature can't be programmed directly into
each single core in the moment that a new core get created

harbaum, 2016-09-09:
Simply because it's some hours of extra work and not all developers have a matching screen and cable to test with. Plus they may simply not be interested in it.

closed 2019-03-25 by lroby74