mist-devel/mist-binaries #32: FPGA graphics not smooth anymore

AmigaIstMiST, 2016-09-24:
- Firmware ATH160910
- C64 v0.27.21

Look at any cracktro textscroller: the graphic is stuttering!
The whole C64 doesn't feel smooth anymore. It feels like an Emulator on an old Windows PC.

harbaum, 2016-10-17:
The scroller inside the on-screen menu stutters? That scrolling doesn't involve the core itself at all. It's the arm io controller drawing into some small video buffer inside the FPGA. But the C64 itself has nothing to do with this.

In that case i assume it's your screen having problems with the C64 video signal and skipping frames.  Can you test on a different screen?

AmigaIstMiST, 2016-10-17:
The scroller inside the FPGA menu AND the C64 itself is stuttering. 
But, as I said above, only the newer versions of the C64 core are affected.
If I use the version fpga_c1541_160110, I have no problems at all, the graphic is completely smooth.

It looks like the video signal of the C64 has changed after this version.

Sorry, I do not have another screen to test against.

harbaum, 2016-10-17:
But if the menu scroller stutters, then it's not the C64 core that has a problem. There are basically two computers running at the same time: The arm io controller doing the on-screen-display and the C64. If both stutter, then it pretty sure the display that's causing the trouble. I'd assume that your screen doesn't like some change in the video timing of the C64 core. Now your screen doesn't properly update and that's why you see both type of scrollers stutter.

Does you screen cope with the video of a real C64?

I'd also assume that the stuttering goes away if you switch between PAL and NTSC (if your screen copes with this at all ...)

AmigaIstMiST, 2016-10-17:
I do not have any old computers to test against, sorry. Old, yellowed computers make me sad, that's why I use a FPGA ;-) 
I tried NTSC-mode, it is stuttering, too.
The question is: was this change in the C64 video signal an intention of the developer (sorgelig?), or is it an unwanted side effect?

EricGus, 2016-10-17:
I don't recall any stuttering on my screen with a crackto screen perhaps its your display? .. can you specify a specific D64 image/prg to try this with i'd be happy to test it on my setup..

AmigaIstMiST, 2016-10-19:
Here it is:

https://drive.google.com/file/d/0B5OPnpPN7u4DMFViYUpucjFOSlk/view?usp=drivesdk

EricGus, 2016-10-19:
Ok got your video but I just want to be sure, the effect you are referring to is that slight "jitter" on the F12 scroll line? ill check a few games on my MiST and see.. but the change Sorgelig did recently was to correct bad NTSC timing (I live in the USA and it was NOT working on actual NTSC displays here)

AmigaIstMiST, 2016-10-19:
Yes, also have a look the green text line with the silly restaurant text, it is visible there, too.

EricGus, 2016-10-19:
Have you tried this without using SCART?

AmigaIstMiST, 2016-10-19:
No, only with SCART so far.

EricGus, 2016-10-19:
do you have a link to the exact D64 image you are using to test with, id like to try it myself on my setup (as well as try OTHER games/disks to see if its a problem with THAT specific image or a problem in general) or with your specific setup/hardware.

AmigaIstMiST, 2016-10-19:
Sorry, I do not know where I got this image from.
It is a general problem for me, I have it on any D64 image and on any scroller. Stunt car racer is only one example. If you can not reproduce problem on your setup so far, I think it may be a waste of time to search for a specific D64 image. I can see the problem even on the C64 start screen, if I look at the blinking cursor.

EricGus, 2016-10-19:
Im still working (its still daytime here) but later tonight when I am home I will give it a try, I just wanted to set up my MiST the same as yours (minus the SCART since its not used here in the USA).

harbaum, 2016-10-19:
I still think it's the screen. Some TVs today implement PAL TV pretty lazy. Some have LCD screens inside which require at least 60 Hz.  So they do some frame repeat to go from 50 Hz PAL input to 60 Hz LCD. If that's done badly the the video stutters. What TV do you use?

AmigaIstMiST, 2016-10-19:
I use DYON TV.

But: I only have this problem with the latest two C64 cores from May/June. The core from January works fine, wonderful soft scrolling. So this can not be a general 50Hz to 60 Hz upscaling problem.The Amiga core works fine, too. (However, the Atari cores (ST,800) do flicker a bit, but differently than the latest C64 cores).
There is definitely a change in the PAL SCART video signal between the January core and the May core of the C64. That was the core, where the NTSC signal was corrected (AFAIK).

EricGus, 2016-10-20:
I am using this version http://csdb.dk/release/?id=13056

I am not seeing the stutter with my current setup, under PAL display setting.

(NTSC setting has some issues with the intro screen -- I'm guessing they never tested it on an NTSC machine)

fpga64 core version v0.27.21
ARM s/w ver. ATH160910
FILE s/w ver. ATH160626

AmigaIstMiST, 2016-10-20:
Here is video using the older core fpga_c1541_160110. The scrolling is completely smooth here:

https://drive.google.com/file/d/0B5OPnpPN7u4DYlFfQmg1RV9LM1k/view?usp=drivesdk

And here again the core fpga_c1541_160627, just to have the two videos in the same comment box:

https://drive.google.com/file/d/0B5OPnpPN7u4DMFViYUpucjFOSlk/view?usp=drivesdk

EricGus, 2016-10-20:
can you try it without SCART .. or try it on a different TV?

AmigaIstMiST, 2016-10-22:
I switched my configuration to a VGA setup and compared the two C64 versions again.
Well, hmmm ....
I cannot see a clearly visible difference in the Stunt Car Cracktro.
But if I open the OSD screen and look at the scrolling MiST info text, I think there is a subtle difference.
I think it stutters less and faster than with SCART, so it's less visible and barely noticable.

sorgelig, 2016-10-22:
You cannot test smoothness by OSD scroller! It's not sync'ed to video at all. It's simple graphics produced by ARM chip and pushed to FPGA. FPGA core simply redraw the video at its own defined rate according to emulated system. So, OSD scroller is totally unsync'ed and not supposed to be smooth by design.

AmigaIstMiST, 2016-10-22:
Well, then this problem is visible on SCART only. I cannot see a difference using a VGA setup.

sorgelig, 2016-10-22:
In VGA mode you have double frame rate, so it may be less visible - it depends on several factors. But in general: OSD scroller is not designed to be smooth or sync'ed to display frame rate.

AmigaIstMiST, 2017-01-02:
It was a problem with the SCART input of my TV. I bought a seperate SCART to HDMI converter and the problem was solved. It looks like my TV can not handle 50hz input very well. After upscaling to 60hz, the stuttering issue with the c64 core went away. For some reason, the only 50hz input my TV could handle was the AMIGA core. All other cores caused problems. Sorry for the noise.

closed 2017-01-02 by AmigaIstMiST