mist-devel/mist-binaries #34: Vic 20 games hung up after last firmware upgrade

lroby74, 2016-11-13:
Hello,
after last firmware upgrade (actually version ATH160910) some Vic20 expanded games doesn't works anymore (my VIC configuration is always 32kb).

A pair of titles are "Critters.prg" and "popeye.prg" (2015 version), both works perfectly on Winvice emulator on my pc.

something is wrong on last firmware about compatibility of games with Vic 20 core?

(I have 2 different Mist, one on SCART and other one with VGA monitor, it's same thing on both units)

Thank you very much
Roberto

harbaum, 2016-11-14:
These games were working before the firmware upgrade? Which firmware did you use before?

lroby74, 2016-11-14:
Yes, i played to them with previous firmware, i don't remember exact
version of previous firmware but was last one before this one

juri74, 2016-11-14:
lroby74 gave me the two files and i tested them with follow firmwares:
160910 = not working
160721 = not working
160626 = not working
one may 2016 release = not working (sorry do not remember what may version i used :P )
160123 = not working
unfortunately i've not time to test ALL firmware releases, i'm not fpga language expert but i suspect the problem is in the vic20 core not the firmware...
maybe some really talented could have a look at the Vic20 core... latest one is TWO years old O_O

harbaum, 2016-11-14:
So one of you is sure that it's a firmware change that broke the vic20 but doesn't know which exact version does work. And the other one claimed it never worked at all.

That'll be hard to debug :-)

Anyway, i'll see what i can do.

lroby74, 2016-11-14:
Honestly i tried Critters only one time and that time has worked (but it
may be only luck)...

from second time it never worked anymore..

i think that Juri74 with his many tried has lighted situation.. problem is
bugged and very old Vic20 core :(

harbaum, 2016-11-14:
The FPGA world is small. Much smaller than e.g  PC emulator world. Thus there are also way less FPGA developers. As a result some things happen slower in FPGA world. But things are getting better with every new MIST owner.

Is there something that the non/bad-working games have in common?

harbaum, 2016-11-14:
I doubt the firmware version has anything to do with this. The firmware has very little impact on the target machine.

lroby74, 2016-11-14:
That i know, a common thing is that both games requires more ram (8K for
Critters and 32kb for Popeye)

lroby74, 2016-11-14:
At this point i think same thing , firmware is not problem.. .core is...
bugged

harbaum, 2016-11-14:
It's not necessarily "bugged". Often it's that a machines hardware has some undocumented side effects then doing things in an unusual way. E.g. on many machines identical portions of ram and rom show up on different places. This is called mirroring and happens when the HW developer tries to save a few gates here and there. And if some software relies on this behaviour then ... boom ...

As i said, i'll have a look.

lroby74, 2016-11-15:
Thank you for you help

lroby74, 2016-12-05:
Hello everybody, some news ? :(

harbaum, 2016-12-12:
I was a little busy the last time. I hope to find some time for this, soon.

lroby74, 2016-12-13:
Don't worry and thank you :)

lroby74, 2017-03-01:
Hello again, sometimes i write here hopefully of good news about Vic 20 core :)

harbaum, 2018-09-27:
The vic20 has vastly been improved and many, many games and even demos now run.

closed 2018-09-27 by harbaum