mist-devel/mist-binaries #35: [Amstrad CPC] core

juri74, 2016-12-18:
not an issue but just a request:
could be possible to "unlock" 15khz RGBS mode please?
thanks

renaudhelias, 2016-12-18:
Amstrad is using a 50Hz display, not interlaced. It cannot be scandoubler for mode 2 (lost of precision)

Using mist.ini scandoubler parameter, you can enable original signal output (SCART), but it was a blind development : some signal are ouput, but VSYNC/HSYNC offset are not confirmed.

I tested with my TV, but no way to run any 15kHz signals on it correctly (SNES doesn't run on it, same problem of display moving up), so I'm blind here. I have to validate a TV before validating my code...

Process generating TV signals is "aZRaEL_process" in
https://github.com/renaudhelias/CoreAmstrad/blob/master/BuildYourOwnZ80Computer/simple_GateArrayInterrupt.vhd
RED_out,GREEN_out, BLUE_out, HSYNC_out, VSYNC_out.

juri74, 2016-12-20:
sorry for late reply,
i noticed other cores uses a RGBS (RGB + COMPOSITE sync) for 15khz video signal, maybe the problem is that HV sync control..
unfortunately VHD is alien language for me :(

harbaum, 2016-12-20:
SCART TVs expect a composite sync. The VGAs vsync is connected the SCART pin used to detect a RGB signal and is constantly driven high. A TV will not cope with a video signal with separate H and V sync.

Bu tit's usually sufficient to xor hsync and vsync to get a csync acceptable for many TVs.

juri74, 2016-12-21:
> But it's usually sufficient to xor hsync and vsync to get a csync acceptable for many 
> TVs.

@renaudhelias maybe could you give it a try?
thanks

renaudhelias, 2017-01-27:
Added to TODO-list
(TODO-list is here : http://www.cpcwiki.eu/index.php/FPGAmstrad#Effort_done)

renaudhelias, 2017-01-27:
http://github.com/mist-devel/mist-board/tree/master/tutorials/soc#lesson-11-15khz-tv-video could be fine to adapt first (50Hz/60Hz), as it I could begin by... validating my TV.

renaudhelias, 2017-01-27:
@harbaum can you align lesson11 60Hz 15kHz with unix "modeline" command protocol ? (refactor) / "modeline" is an unix command showing all world screen/tv parameters for generating easily all world screen/tv signals...

harbaum, 2017-01-27:
@renaudhelias thanks for the hint. I'll have a look.

renaudhelias, 2017-03-08:
I saw in tutorial the use of xor for TV output, so then I try it at home, my TV sucks @15kHz.

So I put a try by here : https://github.com/mist-devel/mist-binaries/tree/master/cores/amstrad/experimental

@juri74 perhaps you can try this not-tested version.

squidgydude, 2017-03-10:
renaudhelias, if you point me to your source code changes for these 15khz tests I can have a look. I've done quite a lot of 15khz work recently and can test it properly.

renaudhelias, 2017-03-10:
Amstrad is 50Hz

Candidate patchs are here (one is inversing the output signal...) :
- http://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/experimental/amstrad_170308_r005.8.14e_xorTV.txt
- http://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/experimental/amstrad_170308_r005.8.14e_xor2TV.txt
Perhaps one between amstrad_170308_r005.8.14e_xorTV.rbf and amstrad_170308_r005.8.14e_xor2TV.rbf is running fine.

Component concerned is :
http://github.com/renaudhelias/CoreAmstrad/blob/master/BuildYourOwnZ80Computer/MiST_SCART.vhd
This MiST_SCART.vhd component does use the mist.ini feature http://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/experimental/mist.ini scandoubler_disable=1

Another component can be patched also :
http://github.com/renaudhelias/CoreAmstrad/blob/master/BuildYourOwnZ80Computer/simple_GateArrayInterrupt.vhd
My second patch is not a great one (xorTV2 and xor2TV2 alternative), I did just try hacking the "HSYNC_out" output (used for TV signal) by here
In this last component simple_GateArrayInterrupt.vhd you'll find the TV signal output generation :
- RED_out : out  STD_LOGIC_VECTOR (5 downto 0);
- GREEN_out : out  STD_LOGIC_VECTOR (5 downto 0);
- BLUE_out : out  STD_LOGIC_VECTOR (5 downto 0);
- HSYNC_out : out STD_logic;
- VSYNC_out : out STD_logic

renaudhelias, 2017-03-10:
If you never tryed CoreAmstrad all is explain here http://github.com/mist-devel/mist-board/wiki/CoreDocAmstrad#installation (you can contact me also : renaudhelias at gmail dot com)

squidgydude, 2017-03-10:
Having a quick look [this one](https://github.com/mist-devel/mist-binaries/blob/master/cores/amstrad/experimental/amstrad_170308_r005.8.14e_xorTV.txt) looks correct to create the csync. I can't check properly until I get home and test the cores and see what signals are being generated.

See my quick conversion changes of TCA2600 [here](https://github.com/wsoltys/TCA2601/commit/c04d3afc82e629f4bd4bbcca1ebb20e31f71d976#diff-7db66cac63872739ce0c098614ab0bba). Specifically:
VGA_HS <= not (vga_hsync_i xor vga_vsync_i) when tv15khz='1' else vga_hsync_i;
VGA_VS <= '1' when tv15khz='1' else vga_vsync_i;

juri74, 2017-03-10:
hello @renaudhelias ,

my wife is at the hospital i do not have time to try at the moment sorry :(
i will publish your test link on my fb page and let other mist users to try it out.
thanks for your effort!!

renaudhelias, 2017-03-22:
amstrad_170308_r005.8.14e_xorTV will be integrated in next realise.
DrOG from atari-forum succeeded in running with it. I think last problem is about VSYNC/HSYNC signal length.
[image]

juri74, 2017-03-23:
great!

renaudhelias, 2017-03-23:
amstrad_170323_r005.8.15c12.rbf published (for ones of atari-forum), normally does resolve OSD in TV mode. It's candidate 12 because I'm currently working on next version (FDC)

KLNHOMEALONE, 2017-03-28:
Please add AMX mouse support if possible

renaudhelias, 2017-03-28:
amstrad_170328_r005.8.15c16e.rbf published "applying Gerald HSYNC/VSYNC width formula"

renaudhelias, 2017-03-28:
@KLNHOMEALONE added to TODO-list (http://www.cpcwiki.eu/index.php/FPGAmstrad)

renaudhelias, 2017-03-31:
"Perfect and well-centered!" - DrOG
[image]

renaudhelias, 2017-04-16:
I've got a TV !
PAL 50Hz 15KHz Parrot test ( https://github.com/mist-devel/mist-board/blob/master/tutorials/soc/lesson11/lesson11.png ) does pass on it, but CoreAmstrad doesn't.
I'll refactor Parrot test into modeline and then in a second step will try to adapt it to CoreAmstrad output :)

renaudhelias, 2017-05-07:
I compare HSYNC/VSYNC width with Parrot test, it results this experimental version : cores/amstrad/amstrad_170507_r005.8.14.2e.rbf (not tested)

renaudhelias, 2017-05-07:
Ok, back from Tetalab laboratory.
[image]
r005.8.14.2e.rbf does run fine, one char or one half-char h-offset in left (screen not horizontally centered)
[image]
All chars are displayed, but HDISP does come too early and go away too later : about one half-char each.
[image]
Ignoring the two previous bugs. Border seems having a 4 chars size (H and V)

renaudhelias, 2017-05-08:
amstrad_170507_r005.8.14.2j.rbf TV mode with borders (not tested)

renaudhelias, 2017-05-10:
r005.8.14.2j OK for me, just an half-char offset of HDISP (cutting chars at right) but border limits OK.
Tryed also with another TV (same model but image not stable verticaly (blurred screen))
[image]
[image]

renaudhelias, 2017-05-11:
r005.8.14.2k , HDISP correction, not tested.

renaudhelias, 2017-05-13:
r005.8.14.2 realized

closed 2017-06-04 by renaudhelias