mist-devel/mist-binaries #37: [OCMSX] Core. Two issues ...

juanfx, 2016-12-28:
Regards, I am having problems with the core 161226. I have a MIST before 2016 and when I boot boot error messages appear in the filesystem msxdos. Pressing F12 shows an incomplete OSD in black and white (MIST style) and pressing again displays the "normal" OSD.

If I go back to the old previous version the problems disappear, I have already formatted the sd card again without result.

Thanks for your time (sorry I had to use the google translator, I am Spanish and I do not speak English at all)

robinsonb5, 2016-12-28:
Thanks for the report.  I've seen some filesystem errors myself, and I'm not sure what's causing them yet.   I will investigate further.

The OCMSX menu responds to both F11 and F12, so you can use F11 to bring up the OCMSX one without the MIST one appearing.  I enabled the MIST one just so that it becomes possible to change the running core.

juri74, 2016-12-29:
@robinsonb5 
could be possible to make the core output video @15khz?
thanks

robinsonb5, 2016-12-29:
@juri74 It can already output 15KHz - you can pick from 31KHz 50Hz, 31KHz 60Hz and 15KHz 60Hz using the OSD.  The OSD can overwrite a config file, but can't create one, so if you want the change to be persistent, put one of the config files from here on the card:  https://github.com/robinsonb5/OneChipMSX/tree/master/Configs
Finally, the 15KHz mode is NTSC. I think PAL is possible, but has to be selected in software on the MSX side.

juri74, 2016-12-29:
thank you very much!

juri74, 2016-12-30:
hello i'm running msx core in 15khz mode thanks however i noticed a bug, when i open MIST menu to change core, the menu is 80% displayed off the screen, to the right and it is unusable.
picture:
https://www.facebook.com/photo.php?fbid=10211489983836444&l=5d4d4ef987

harbaum, 2016-12-30:
Looks like the OSD pixel needs to be adjusted. Usually using half the pixel clock in 16khz mode is sufficient.

gyurco, 2019-11-18:
I think these issues are not in the current core.

closed 2019-11-18 by gyurco