mist-devel/mist-binaries #41: [ECS Minimig] 15 kHz mode not working

AmigaIstMiST, 2017-04-07:
The scandoubler_disable option doesn't work on original minimig core.

- minimig core r246
- firmware ATH170207

With the same MIST.INI file, the MinimigAGA core is booting in 15kHz mode without problems.

rkrajnc, 2017-04-10:
The old ECS minimig core does not support any setting from mist.ini. It shouldn't be a lot of work to make that happen, though - just updating the user_io.v module and connecting a wire.

Nice project for someone starting in Verilog, perhaps?

AmigaIstMiST, 2017-04-10:
Ah, OK. I thought that feature was already available in the ECS core and was broken. I didn't know it was introduced in the AGA core.

Sorry, I am a software guy, I have no clue about about hardware. A little testing and bug reporting is all I can do. I am interested in the Amiga, not hardware in general, and the Amiga AGA core is nearly done anyway :-) .

AmigaIstMiST, 2017-04-10:
I think I leave this ticket open, maybe it attracts someones interest to implement this feature ...