mist-devel/mist-binaries #43: [Amstrad CPC] Commando

bonnefoi, 2017-04-24:
Hello,

Do you think that fixing the Commando screen artefacts could be possible ?

best regards,
Pierre-François.

renaudhelias, 2017-04-24:
Commando screen is atefacts seems due to Z80 implementation (instruction speed, instruction quality). I have a Z80 testbench, and it does fail a lot (but it is my favourite Z80... ^^')

bonnefoi, 2017-04-24:
Do you have taken a look at the A-Z80 core by Goran Devic ? You have already succeeded to use external memory so perhaps you could try another core... :)

sorgelig, 2017-04-24:
A-Z80 is more far from Z80 replica than T80.
I've started my ZX core with A-Z80 and later found many problems. Some instructions aren't correctly implemented. Timings are also not correct. Author of A-Z80 claimed it's based on chip decaping, but i think it's not true according to bugs i've found.

renaudhelias, 2017-04-29:
A-Z80 was allready tested on CoreAmstrad, it is worse.
That is certainly due to the fact I did progress my project with T80.
But if it is worse, so then it is "different".
That's a interesting fact, A-Z80 is using a different approach than T80.
T80 is a translate from MAME one if I know enough the author of it.
A-Z80 seems a full-hardware approach, using low level set of components as a gallery of ants, and ... does run.
Facing each other shall be a great and cool way to progress.

renaudhelias, 2017-04-29:
Let's, for the fun, try Commando with A-Z80.
I turn on the parameter : USE_AZ80:boolean:=false; in FPGAmstrad_amstrad_motherboard.vhd
Compiling...

renaudhelias, 2017-04-29:
Testing...

renaudhelias, 2017-04-29:
Does boot...

renaudhelias, 2017-04-29:
CRTC:1/MEM_wr:quick : KO (normal quick is not original)
CRTC:1/MEM_wr:slow : KO
CRTC:0/MEM_wr:slow : KO
CRTC:0/MEM_wr:quick : KO
Exactly as on CoreAmstrad (damn !)

Game test :
-Moktar : KO - some problems while reaching bottles (this was my best friend favorite game...)
-Star Sabre (a 2009 spanish game) : OK
**-Sim City : INTRO HACK DEMO DOES PASS PERFECTLY IN A-Z80 and NOT WITH T80 (2/3 of luck)**
-Edge Grinder (squeeze screen bug in CoreAmstrad) KO, same
-Pinball Dreams (recent game maked by Batman group !) - freeze with MEM_wr=slow, DOES RUN WITH MEM_wr=quick (as in CoreAmstrad)

OK, next challenge, DEMO (they are known as testbench booting, detecting and rage quit if something is bad emulated... forcing as it buying the original CPC ^^). In CRTC1/MEM_wr:slow
30YMD (not running so fine in CoreAmstrad, and in a lot of emulators) : rage reset, directly.
the legendary Batman Group's demo (running not so badly in CoreAmstrad) : first part : PERFECT sound/PERFECT graphics second part : quite GOOD
MEGABLASTERS death (in CoreAmstrad it does run)
Points barres : first part OK, second part KO (as in CoreAmstrad) crash.
Midline Progress (not running in CoreAmstrad) KO (normal it is made by god itself : ARKOS, certainly a FDC bug) terrible crash, even the reset button of MiST does nothing, and received an electric choc touching the platform.
From Scratch (oh oh oh) does crash (as in CoreAmstrad damn it !)

And MUSIC (cause CPC has a Yamaha chip, respect) :
STARKOS (by ARKOS) - the CPC best soundtracker ever, CARPET.SKS => PERFECT

And ROMs :
PROTEXT ok

renaudhelias, 2017-04-29:
TODO-list updated (http://www.cpcwiki.eu/index.php/FPGAmstrad) just do search "A-Z80" on this manifest.
Good job, well done, a lot of thanks to you guys !

renaudhelias, 2017-04-29:
¤ in fact, small electric choc comes from my speakers...

renaudhelias, 2017-04-29:
Can be closed, commando is already in TODO-list.

renaudhelias, 2017-04-29:
do play Ikari Warriors instead ^^

bonnefoi, 2017-04-29:
Ikari warriors :D it's too damn hard for me !
thank you for trying with A-Z80 👍

sorgelig, 2017-04-30:
The most important (and sad) fact is: there is no Verilog/VHDL exact replica of Z80...  All Z80 modules are based on known functional model and don't include very specific less documented nuances.

closed 2017-05-13 by renaudhelias