mist-devel/mist-binaries #47: Networking?

bluecursor, 2017-08-30:
Add the ability to use ether usb ethernet stick or usb wifi dongle in fpga64 and minimig.

harbaum, 2017-08-30:
Can you elaborate which part is supposed to deal with the network specific settings? Where is the network stack (tcp/ip and friends) supposed to run? On the C64?

bluecursor, 2017-08-30:
I am not a tech guy, just a end user.
I don't know specifics but on the Amiga, would having installed Amitcp or other type of network stacks work as it currently works on a real Amiga with a pcmcia wifi card ?

I am unsure how they currently do it with a real64 with these new wifi cards

I would just like to have the ability to use my mist fpga when using either c64 or Amiga cores to have use of the internet. Since the mist fpga doesn't a serial port, I assume using a USB Ethernet stick would be suitable.

Again, I am. Just a user.

_o/ Sent with iOS Spark

smokku, 2017-08-30:
- The de-facto standard for ethernet interface for C64 is [RR-NET](http://wiki.icomp.de/wiki/RR-Net). Other ethernet cartridges for C64 (like ETH64) are RR-NET compatible.
- TCP/IP stack for C64 is [IP65](http://oliverschmidt.github.io/ip65/).

harbaum, 2017-08-31:
I have actually built a custom TFE cartridge for my SX64 many years ago. See http://harbaum.org/till/c64/index.shtml While this physically works it's imho not much more than a funny tech demo worth a few minutes of fun. THere's no real use case for this.

It is possible to implement something in VHDL that behavs like the chip used on the RR-NET or the TFE. But that's a few weeks work and it wouldn't work for WLAN USB dongles. That would need a seperate user interface on the arm side of the mist board to configure the wireless specific stuff.

EricGus, 2017-08-31:
What would be REALLY useful is to implement a wifi modem like interface instead of an RR-NET, this would allow you to use regular C64 telecom software to connect to internet BBS's in operation today (thats how most c64/128 users are getting online these days), (RR-NET and direct TCPIP software on the 64 has limited options) on the other hand, the Wifi Modems currently utilize an ESP8266 serial to wifi chip and replicate all the functions of the old HAYES modems (for the Commodore this would be equivalent to the 1670), so there is wide support program wise that can utilize it (pretty much all the old programs). The ESP8266 software basically just runs TCPSER on the chip .. Some people have on the Mist connected the midi RX/TX lines under the Amiga core to these chips for serial comms, but as we know the FPGA64 doesn't enable them so this doesn't work. It would either be nice to have the RX/TX midi lines on the mist enabled for the FPGA64 core, or somehow replicate the functions of the Wimodem/TCPSER software on the ARM in the Mist using a usb wifi/network dongle. The ESP8266 code can be found here https://github.com/RolandJuno/esp8266_modem

gyurco, 2019-05-02:
RX/TX lines are usable now for an UART in the C64 core. Tested successfully with a PL2303 serial-USB adapter.

boomlinde, 2021-10-04:
Are there any plans to add an Ethernet driver to the Minimig core? I recall seeing it considered before, but since then there's been some interesting prior art, e.g. [the A314 SANA2 network driver](https://github.com/niklasekstrom/a314/blob/master/Software/ethernet/device.c) which I think might be useful as a reference. It seems to work by passing messages to and from a computer running Linux.

gyurco, 2021-10-05:
It's still in the planning phase :)

flewid, 2023-05-17:
Hello, I am also wondering the status of getting networking working with Minimig? I have the ASIX AX88772 and it's detected fine by the menu, but I can't figure out what driver to use on Amiga 1200 using WB 3.2.2.1 - roadshow installed fine, as did ibrowse + ssllib and mui, but I just can't get roadshow to add the device - I'm assuming a driver is missing to 'see it' as an interface?

flewid, 2023-05-17:
Darn was hoping that wasn't the case. Is it difficult to add? Would this work if I got a USB to serial? - https://misterfpga.org/viewtopic.php?f=4&t=2063&p=18598&hilit=Roadshow#p18598

gyurco, 2023-05-17:
You can use the MiST's UART port (or the MIDI in/out ports) for PPP directly. Serial redirection to USB is not implemented for Minimig.

gyurco, 2023-05-17:
https://hackaday.com/2019/07/03/amiga-in-the-mist-gets-online-with-an-esp8266/

flewid, 2023-05-17:
nice thank you!