mist-devel/mist-binaries #49: [CoreDocApple2 / PlusToo] 15khz support

aixtreem, 2017-11-09:
Hello,
To the respective core developpers: Is it possible to add 15khz support ?
would be great as many users run on CRT monitors

thanks!

harbaum, 2017-11-10:
CRT screens don't necessarily need 15khz. In fact the vast majority of old VGA screens doesn't support 15khz. Only old TVs and those C64/Atari ST/Amiga screens do.

Anyway, adding 15 khz to the PlusToo isn't possible as the MAC itself didn't support that video format back in the days. The Apple2 on the other hand could likely support 15khz.

aixtreem, 2017-11-10:
thanks for your reply, didn't know for the macintosh.
and yes, sorry, CRT is philips/tv/amiga/atari monitors in my mind :)

closed 2019-10-20 by gyurco