mist-devel/mist-binaries #51: [Minimig-AGA 1.2.1 core]  Reboots with USB keyboard

lexthehex, 2018-02-17:
This has been a complex bug to reproduce and I'm not sure of the culprit. 
A description of the symptoms: MiST resets when the keyboard Enter key is pressed, upon boot.
I have chip RAM set to 1.0 MB, slow set to 512K and Fast RAM to 2.0MB. I am using Kickstart 2.05.  The issue always occurs with the AGA chipset selected, sometimes with ECS and never with OCS.   A description of the bug: when pressing Enter on the USB keyboard, MiST pauses and then resets. When this is about to happen, the F12 menu also sometimes does not appear and takes a few presses of F12 before it shows.  The issue occurs more if you press the Return key.
I'm using an HDF file to boot Workbench, and am then running WHDLoad to play games.  The scandoubler is disabled so that I can use MiST with my TV.

lexthehex, 2018-04-16:
To renaudhelias: thank you for your reply.  Yes, it's very likely that my MiST PSU is not able to supply enough current - I'll buy a new one.  Closing this issue.

closed 2018-04-16 by lexthehex

harbaum, 2018-04-16:
Renaud is probably right. You might also just use a different keyboard or mouse. Maybe the ones you are using are drawing unexpected high current.