mist-devel/mist-binaries #52: scrolling OSD message does sucks

renaudhelias, 2018-04-13:
Only a newbie has to read the fucking manual in life.
Please all newbies by here, please read this sentence, you HAVE to BUY a JOYSTICK in order to HAVE FUN with all reborn here games... and have to buy pleaaassee a great old school one -like the pro 
 competition joystick-.

And please do continuous building MiST-board board for us because I've got a lot of work in next years on it :P Amstrad !

renaudhelias, 2018-04-13:
And if another want to contact me about...
* making a FPGA DAC dog ear compatible, so fine that while playing a duck gun game some ducks does come bottom front of our window...
* ... and stupid thing like displaying a VGA output 72Hz 800x600  display using FPGA VRAM...

harbaum, 2018-04-16:
This is a community project and you have write access to the repositories :-)

Actually these scrolling texts were there in the Minimig and i removed them for the MIST as i also didn't enjoy them too much. Some re-inserted them which is totally fine as i am not the one to decide such things alone.

closed 2021-07-16 by squidrpi