mist-devel/mist-binaries #57: Sorgelig please don't abandon Mist support :(

lroby74, 2019-01-14:
Hello Sorgelig,
sorry if i write you in this way but i don't know how i can do it in other ways..

I am asking you about don't abandon Mist support, on updatings his cores..

I love this device a lot and i am very afraid that without you, support for this one may die

Thank you very much and forgive me
Best regards from Italy
Roberto Lari

jotego, 2019-01-14:
I totally agree. We miss @sorgelig magic!

sorgelig, 2019-01-14:
You have Gyurco who is very active on MiST. We often cooperate in cores improvements.
Jotego also can do much more than pretend :)

jotego, 2019-01-14:
Ok. Ok. I'll release one core for MiST this week!

harbaum, 2019-01-15:
That's great.

Anyway, there are many people who might be asked. E.g .the makers of the mistica and the mini st clones have imho never contributed anything. It may make sense to ask them as this would of course be for their own benefit.

gyurco, 2019-01-29:
But mistica is mostly unusable for development :)

jotego, 2019-01-30:
Why is mística unusable for development?

gyurco, 2019-01-30:
Does it have a JTAG connector?

lroby74, 2019-01-30:
I don' know that

jotego, 2019-01-30:
I checked with the author of the board. No, it doesn't have JTAG
connectors. Good for users anyway!

gyurco, 2019-01-30:
But imagine...no MiST will sell because of cheap clones without JTAG. You cannot get a MiST after a while. No developer can join. MiST dead.

sorgelig, 2019-01-30:
Not every dev use JTAG, actually.
Jotego got JTAG just couple days ago. 
The author of SNES also didn't use JTAG till may be 1 month ago.

gyurco, 2019-01-30:
Jotego is professional, he can write simulations for everything :)
My personal experience that using USB-Blaster is like drilling holes to the wall with hand or an electric drill.

closed 2019-04-24 by lroby74