mist-devel/mist-binaries #59: C16 and SID

Rusazar, 2019-03-03:
Greetings
Is it possible to add SID support to C16?

harbaum, 2019-03-03:
What software would be using that?

Rusazar, 2019-03-03:
Here are recent examples presented in demomaking:

http://www.pouet.net/prod.php?which=76092 - Waiting For Promised Land
http://www.pouet.net/prod.php?which=77695 - Promised Land

Perhaps there are games that can use the SID card. SID card connection by ports is described here:

http://bsz.amigaspirit.hu/nae/NST_SIDcard-20140607-En.pdf

I did not find the scheme, there is no connection with the author. I would like to fully implement the C16 in order to fully get the impression from the use of SID.
Thank you for your cooperation and possible implementation in the FPGA project.

harbaum, 2019-03-05:
There are only two demos using this? I doubt it's worth implementing this in the main core as 99.9% of all users don't have a use for that.

One of the nice things of the MIST board and it's cores is that they are not totally feature loaded and they don't come with settings which confuse most of the users. 

You can of course build a "hacker edition" with as many features as possible. This actually seems to be a perfect project for a FPGA beginner as you likely only have to wire the parts up correctly.

Rusazar, 2019-03-05:
No, you didn't get me right. I gave You only 2 demo for example. Such works are much more than 2 pieces. I can still give you a lot of such works, where for C16 will be used SID card. 

This is the standard for users of the C16 platform. And when you look at MiST such demos, it's the same as watching a silent movie... I have no experience with FPGA and I can't improve the C15 kernel, add SID card to it. Therefore, I address this request to the developers. 

Please integrate the SID card into the C16 kernel.

harbaum, 2019-03-05:
None of us was an FPGA developer when we started working on the MIST. The most important thing is that everyone contributes and there's no distinction between developers and non-developers.

Please make yourself familiar with the tools and give it a try. It e.g. seems that there are various ways to access the SID and that there are different compatibility modes which are supposed to be implemented correctly to make a kinds of demos work. It even seems that there are demos backported from the c64 which generate SID sounds on a C16 if the SID is mapped the same way it's mapped on the C64. On the other hand this requires some special reset logic since this may produce accidential noises with software that is not aware of the SID.

So why not learn something new?

Rusazar, 2019-03-05:
New is always good, I do not argue. Development is our life. 

But who will find it easier to implement improvements in the C16 kernel? I am unfamiliar with this subject or a person who is already easily oriented in VHDL? How much time will I spend on implementation or the same specialist? 

The answer will be obvious. I do not force anyone to do anything, I only ask, if there is time, the ability to implement in the core of C16 external sound chip-SID. With it, this core will be complete and people who like to watch demos will watch them with sound, not in silence...

harbaum, 2019-03-05:
You just gave the perfect reason to join in: If you don't do it then there's a chance noone will ever do it. So it's not the question if someone _can_ do it faster than you but if someone actually _does_ it faster than you.

gyurco, 2019-03-06:
Answer to the original question:
Is it possible to add SID support to C16?
Yes :)
I think no one will reject a pull request to mist-devel/mist-board/cores with that.

Rusazar, 2019-03-06:
That would be great, I look forward to it!

gyurco, 2019-03-06:
I didn't say I'll do it :) At least not soon.

Rusazar, 2019-03-06:
It will be good that there is any hope that this will be done at all. Even that answer is encouraging. And for terms of implementation nobody speaks. How will personal the opportunity to do so, it will be a nice bonus for all the fans of C16.

I personally thank you for at least a hint that it is possible to implement sid support in the C16 kernel. It's better than nothing at all. ;)

gyurco, 2019-03-06:
This work would be more about researching how the SID is connected to the Plus4 on the expansion card, not about the actual core/Verilog implementation. It's very easy if you know how the hardware works. That's why @harbaum wrote it would be a perfect opportunity to jump in to the development.

Rusazar, 2019-03-06:
Here's what I found on the circuitry SID card:
http://solder-synergy.de/plus4/hardware/makers/sid.gif

It was taken from this source:
http://solder-synergy.de/plus4/hardware/index.html

Perhaps I will contribute to the implementation of this project.

Rusazar, 2019-03-06:
And here is a list of programs using SID card, I think it is complete. And as I see in this list clearly much more 2 programs. :)

http://plus4world.powweb.com/effects/SID_support

harbaum, 2019-03-06:
Are the equations of the gal available? We need to understand what the gal is supposed to do.

smokku, 2019-03-06:
Are you asking for the equations in the centre of the .gif ?

gyurco, 2019-03-07:
And the extra joystick port is used by games?

harbaum, 2019-03-07:
> Are you asking for the equations in the centre of the .gif ?

Uhm ... yes :-)

Rusazar, 2019-03-07:
> Are the equations of the gal available? We need to understand what the gal is supposed to do.
https://www.farnell.com/datasheets/7379.pdf

sorgelig, 2019-03-07:
Some games are from 1993, 1994.. I doubt SID addon was so common at that time. It's more modern addition. 
So that list of games supporting the SID is doubtful.

Rusazar, 2019-03-07:
> Some games are from 1993, 1994.. I doubt SID addon was so common at that time. It's more modern addition.
So that list of games supporting the SID is doubtful.

http://plus4world.powweb.com/effects/SID_support
This link contains the full list of programs released for C16, in the amount of 274 pieces. The first program "Future Composer V2.1 + SID" is dated 1992-09-12.
I think that the SID card implementation in the FPGA C16 core will not be superfluous.

gyurco, 2019-03-07:
At the end it's a very easy addon, just decode the address bus to FD40-FD5F and FE80-FE9F for SID. 
Another question, which SID model should be used? 8580 or the old one? Or both? Stereo option? Somebody who really wants this should really try to add it :)

Rusazar, 2019-03-07:
Is it a problem to make a choice in the settings menu between the FD40-FD5F and FE80-FE9F addresses for the SID?

And also make a choice for the SID model? 

It's just a question, I just don’t know if it’s difficult or difficult to make a choice of configurations, is there enough FPGA cells for this or not enough?

harbaum, 2019-03-07:
Ok. So here's the offer: The op does it and if he runs into problems I promise tp help him.

sebdel, 2019-03-07:
I don't know if you're serious about this but Till tutorials are pretty amazing to get you started: [https://github.com/mist-devel/mist-board/tree/master/tutorials/](https://github.com/mist-devel/mist-board/tree/master/tutorials/)
They teach you enough to get some meaningful things done and get a bit overconfident and then you run into timing issues and you realize just how ignorant you really are. It's a blast.

harbaum, 2019-03-07:
Since the possibility to choose is already present in the c64 core it will very likely also fit as part of the c16 core. Also this shows how this can be done. Imho the c64 also supports a stereo dual aid setup. This does not exist for the c16, does it?

piso77, 2019-03-07:
[OT] Ok, i don't own a Mist, but i received a Mister this week (still packaged though...) and i know a bit of VHDL, and while the C16 core is Verilog, i would still like to give it a shot - bear with me...

gyurco, 2019-05-03:
SID added

closed 2019-05-03 by gyurco