mist-devel/mist-binaries #60: firmware_190303  breaks "load config"

theflynn49, 2019-03-04:
Hello ...
This firmware release breaks the "load configuration" function from minimig-AGA 1.2.3 
symptom : once a load configuration id done, OSD will not come up any more
replicates exactly #24 
reverting to firmware_190101 fixes the problem.
Hope this helps. 
Rgds

gyurco, 2019-03-06:
I don't see how the recent changes could affect that...will check.

gyurco, 2019-03-06:
Tried it, it works for me. Load configuration reloads the core, during the reload the OSD is not available. But that's only 4-5 seconds.

theflynn49, 2019-03-06:
Thank you for your time and your support.
Due to the other ticket, that called for a regression.
I reinstalled firmware_190303 to try it with other minimig cores : 1.1.0, 1.2.1 and 1.2.3; they all 
behave the same way : loading parameters resets the core all right, but thereafter no way to get the OSD up any more.

I guess it's a weird timing with my mist hardware then. Let me close this ticket for now.

theflynn49, 2019-03-14:
I just worked on this issue. I hooked a serial console and a JTAG debugger on the arm proc.
firmware_190303 breaks right in the middle of writing the console message "CPU clock : ..." during the load configuration command after the core is resetted. It doesn't make any sense unless it misbehaves during an interrupt or something.

Anyway I downloaded and recompiled the firmware, and my local version works fine.
So I don't know what to do with this, it may help others to know that the image firmware_190303.upg has issues for me.

gyurco, 2019-03-14:
Great you can worked it around. Actually sometimes I see different problems with different compilers with the firmware (e.g. if I compile with one version of GCC+newlib, the sort order of the files will broken), so can I ask what compiler+newlib did you use?

theflynn49, 2019-03-14:
(sorry for the delay - internet connection problems :( )
arm-none-eabi-gcc -v
gcc version 6.3.1 20170620 (15:6.3.1+svn253039-1build1) 
dpkg -l | grep newlib
ii  libnewlib-arm-none-eabi                         2.4.0.20160527-3 
ii  libstdc++-arm-none-eabi-newlib                  15:6.3.1+svn253039-1+10
 cat /usr/include/newlib/_newlib_version.h
#define _NEWLIB_VERSION "2.4.0"

harbaum, 2019-03-14:
Freel free to release your better working version. There's one thing you should test beforhand: Update to the new version and then test it's self-update capability by downgrading again. If that routine is broken then people may have a hard time updating later.

theflynn49, 2019-03-14:
Ok, 
- I recompiled the sources from a fresh clone from the repo to make sure I didn't made any change to it
- I checked that it still fixes this issue
- I checked the downgrade upgrade and ...
- I just made a pull request :D
thanx

closed 2019-03-22 by theflynn49