mist-devel/mist-binaries #62: Videopac - Gunfighter game doesn't run on core

Gerloc, 2019-04-21:
The Game Gunfighter (name in eu) Showdown in 2100 AD (name in usa), Dos not run on core.
Game freeze about 1 sec after loaded in core.

This is one of the most popular games for the Videopac Console. 
Most other games seems to work though i have not tested them all.
Hope to see a fix for this.

The Videopac Console had some AddOns, like the Videpac plus console that for some games could show more Advanced backround Graphics, Also the voice module that enabled synthesized Speech in some games..

Gerloc, 2019-11-23:
I testet this Videopac Core some more, and there are several other games with some problems and issues on this core. The sound is also not accurate. It was Nice to see the addition of new standard menu and that the core now runs on 15 hz display. The emulation core itself is still rough and not accurate.  Hope to see some improvement on this core. Thanks for the excellent work on many of the other Mist cores..

gyurco, 2019-11-23:
That would need a developer who's interested in this hardware. Seems currently no one is.

Gerloc, 2019-11-24:
Yes i see Your point. There are some interest in the Videopac console in some retro communities, So hopefully it will be looked into sometime in the future...

closed 2020-01-13 by Gerloc