mist-devel/mist-binaries #63: Pcengine (TurboGfx16) core does not work keyboard or gamepad

Stewings, 2019-05-17:
Hi,

Everything *.pce rom load OK besides does not respond joystick or gamepad or keyboard. Github’s description does not include any keyboard layout. The question would be, how can I use the games and what do I do?

sorgelig, 2019-05-17:
assign the gamepad/keyboard buttons in PCE core.

lroby74, 2019-05-17:
You can simulate a Gamepad via keyboard pressing NUMLOCK sometimes until when Windows Logo, Left Alt, Left Ctrl & Left Shift Keys becoming active (Numlock cycles joystick 1, joystick 2, mouse and nothing)

Stewings, 2019-05-17:
Hi,

It’s OK. Missing my keyboard NumLock. I changed my keyboard now. I try your description!

Thank you for your help!

Best Regard,
Steve

> 
> You can simulate a Gamepad via keyboard pressing NUMLOCK sometimes until when Windows Logo, Left Alt, Left Ctrl & Left Shift Keys becoming active (Numlock cycles joystick 1, joystick 2, mouse and nothing)

Fekete István  <http://www.linuxadmin.hu/>
Linux Supervisor DevOp 

www.linuxadmin.hu  <http://www.linuxadmin.hu/> |  info@linuxadmin.hu  <mailto:info@linuxadmin.hu>
M: +3630-351-2048
C: 1149 Budapest, Nagy Lajos Király u 109/a.
1.emelet 3.
.
T: +36 1 787 33 08 |		linuxadmin.hu 
Linux adminsztrációs portál <http://www.linuxadmin.hu/>

sorgelig, 2019-05-17:
No need to press anything. Just define the buttons as i've told.

lroby74, 2019-05-17:
> No need to press anything. Just define the buttons as i've told.

On Mist too?

sorgelig, 2019-05-17:
Sorry.. Worked too much. 
I've though it's MiSTer repo...

gyurco, 2019-05-20:
Which core are you using?
This is the most recent one:
https://github.com/gyurco/fpgapce
It doesn't respond to keyboard, since it only uses joysticks. If joysticks don't work, probably you have an ancient firmware on your MiST.

Stewings, 2019-05-22:
Hi,

My core version is: 

Lotharek mist-devel latest version:

https://github.com/mist-devel/mist-binaries/tree/master/cores/pcengine <https://github.com/mist-devel/mist-binaries/tree/master/cores/pcengine>

https://github.com/gyurco/FPGAPCE/releases <https://github.com/gyurco/FPGAPCE/releases>

https://github.com/gyurco/FPGAPCE/releases/tag/190215 <https://github.com/gyurco/FPGAPCE/releases/tag/190215>

I tryed change F12 submenu joystick settings and I founded swap joystick enable. This setting is half OK. My pc keyboard SHIFT, Control, Numlock and UP,DOWN,LEFT,RIGHT keys is live. The joystick is half live all direction OK however only second Fire button working. First fire button only keyboard available.

Sorry my answer too late. I am very busy. Lot of work.

Best Regards,
Steve  

> 
> Which core are you using?
> This is the most recent one:
> https://github.com/gyurco/fpgapce <https://github.com/gyurco/fpgapce>
> It doesn't respond to keyboard, since it only uses joysticks. If joysticks don't work, probably you have an ancient firmware on your MiST.

Fekete István  <http://www.linuxadmin.hu/>
Linux Supervisor DevOp 

www.linuxadmin.hu  <http://www.linuxadmin.hu/> |  info@linuxadmin.hu  <mailto:info@linuxadmin.hu>
M: +3630-351-2048
C: 1149 Budapest, Nagy Lajos Király u 109/a.
1.emelet 3.
.
T: +36 1 787 33 08 |		linuxadmin.hu 
Linux adminsztrációs portál <http://www.linuxadmin.hu/>

gyurco, 2019-05-22:
It's possible that button A and B are swapped in the core, but in Street Fighter II options screen, it seems to be correct. You can test on the controller setup menu in SF2 what button it sees on your joystick. Btw, most games on PCE require two buttons, and a 1-button joystick will be useless anyway.

gyurco, 2019-05-25:
Well, here are the various original controllers used for the PCE:
http://www.pc-engine.co.uk/?section=controllers
As you can see, button I and II are counted from right-to-left, and probably that's why most games are using button II as the main fire button.

closed 2019-10-20 by gyurco