mist-devel/mist-binaries #67: Amstrad Core (Legend of Steel) Graphical issues

micv2, 2019-11-22:
Hello,
a nice game, unfortunately graphic error at 50 and 72 Hz. 
My monitor can be 50Hz. 48Hz not. 
Why does the game show 48Hz on my monitor?

https://cpcrulez.fr/GamesTest/legend_of_steel.htm

Best regards, MiC

renaudhelias, 2019-11-22:
You can try also the scandoubler (menu VGA: scandb50Hz)

renaudhelias, 2019-11-22:
Added to http://www.cpcwiki.eu/index.php/FPGAmstrad TODO-list

renaudhelias, 2019-11-22:
Does run fine on Sorgelig core fork.

micv2, 2019-11-22:
menu VGA: scandb50Hz.. cover picture is Ok, then monitor outside spec.

[image]

[image]

micv2, 2019-11-22:
At 72Hz the game graphics is broken.

[image]

micv2, 2019-11-22:
Other games work very well. Ok, I would also like scanlines at 50Hz. 
This is the first time I see a game at 48.8Hz?! 
What is the Amiga with interlace mode? ;)