mist-devel/mist-binaries #68: NES core seems to have display position to mutch to the left at 15 khz.

Gerloc, 2020-01-02:
Noticed that on a 15 hz crt screen the image on the NES Core seems to have the display position to mutch to the left in comparison to most other cores.  Even though i can shift the position with my crt screen to the right somewhat , Even then the NES core is not complete  in center and still about 1 cm black bar is visible on the right side of screen.  Im using the scart cable with a 14 crt tv.

Gerloc, 2020-01-02:
I would like to add that its not a overscan issue, If i put on crop overscan in NES core then the display is like 1 cm black bar on the left and about 3 cm black bar on the right. Still not in center.  If i run the GENESIS core and have the position dead center and the run the NES core without adjusting anything, The NES display is of center to the left about 4 cm aprox..  Anyone else have seen this issue on 15 hz screen ?

Gerloc, 2020-01-10:
Maybe the mist firmware could have a x position and y position adjustments for 15 hz if thats possible ? So that issues like these can be avoided for all cores ? Since the different cores seems to have variable center positions in 15 hz. And the NES cores seems to be way of to the left, not able to center at 15 hz.....in my setup. Others may have other experiences.

gyurco, 2020-01-10:
It's not possible to do it in the firmware. The cores are generating sync signals on their own. I've updated the NES sources last year, but I didn't test it too much. You can try it, if you can compile.

Gerloc, 2020-01-11:
I would like to to try that. At the moment i don't have the knowledge on how to compile the Mist cores, But i would like to the some Research on that matter.. thank you.

closed 2020-01-13 by Gerloc

Gerloc, 2020-02-22:
This issue was now solved by the latest Nes Core realease..:)