mist-devel/mist-binaries #69: Minimig-AGA 15khz VGA

Angel-child, 2020-01-10:
Hello, is it possible to get an 15khz VGA Signal from MINIMIG-AGA?
Best regards Torsten.

lroby74, 2020-01-10:
I think that VGA works only on 31khz.. 15Khz is for classic TV or  CRT monitors pre VGA

Angel-child, 2020-01-10:
Yes Thats why i want it.... for an classic 15khz VGA Monitor

lroby74, 2020-01-10:
if i remember correctly, only Multisync VGA monitors can use 15Khz

Angel-child, 2020-01-10:
Yes Thats right, but I have such ones... (use it with an Amiga 3000). And the picture is really much better with 15khz, that's why i asked if its possible to get also an 15khz in minimig-aga.

gyurco, 2020-01-10:
Yes, it's possible, disable the scandoubler in mist.ini or hold the middle button for some seconds.

Angel-child, 2020-01-10:
But if i disable the scandoubler in mist.ini it gives me a csync Signal... no hsync c and vsync...
Ohhhh can it implemented? 🙂 I think its an nice option to select 15 or 31 kHz output for VGA.

gyurco, 2020-01-10:
It's just the matter of changing these two lines:
https://github.com/mist-devel/minimig-mist/blob/master/rtl/mist/video_mixer.sv#L95
But would be good to have a common configuration method in the firmware to make it a general solution (maybe an option in mist.ini).

harbaum, 2020-01-10:
Did you actually test it with csync? The few screens I have that cope with 15khz also cope with csync incl. a NEC TFT PC Display.

Angel-child, 2020-01-10:
Yes i have tested it... with my Monitors it didnt work with the csync Signal. I use also an NEC.. . but I didnt get a Signal with the csync.

Angel-child, 2020-01-11:
> It's just the matter of changing these two lines:
> https://github.com/mist-devel/minimig-mist/blob/master/rtl/mist/video_mixer.sv#L95
> But would be good to have a common configuration method in the firmware to make it a general solution (maybe an option in mist.ini).

That looks like i wish about.... but iam not able to do this, iam only a User since many years, not a programming genius like you 😉

gyurco, 2020-01-12:
I've added a csync_disable option to the firmware (can be used in mist.ini), and corresponding change to the minimig-aga core. Next versions will include it, or you can compile now for yourself as always.

Angel-child, 2020-01-12:
Thank you so much... Cant await to test it 🙂

Angel-child, 2020-01-17:
Hello, please is anyone able to compile me to New Firmware sourcecode so i cant test it? 😔 Since one week i try the GCC Compiler, nothing works. The "make" only bring me error error error.... Ive used the installscript, i intalled the libraries all like in description... I cant get it to compile since one week of trying. I give up now with the GCC.  I despair 😭
[image]

sebdel, 2020-01-17:
It just works with a docker env for me. Try this:

If you don't have docker installed, do it:
`sudo apt install docker.io`

Then go to the firmware directory and do:
`docker run --rm -v $(pwd):/build st3ph4n/gcc-arm-none-eabi make`

Hope that helps :)

Angel-child, 2020-01-17:
> It just works with a docker env for me. Try this:
> 
> If you don't have docker installed, do it:
> `sudo apt install docker.io`
> 
> Then go to the firmware directory and do:
> `docker run --rm -v $(pwd):/build st3ph4n/gcc-arm-none-eabi make`
> 
> Hope that helps :)

Oh Ok, i will test this after my work. 🙂 Thank you

Angel-child, 2020-01-17:
Have the solution : its the makefile.... The path in the First line... Now i have the New Firmware compiled 🙂
But the New core is the next step 😂 dont know if i can do it, it sounds much more complicated. Perhaps i wait for the Release, hopefully not so long.

gyurco, 2020-01-17:
The most complicated thing in core compiling is registering to Intel and downloading ~2GB of data.

harbaum, 2020-01-17:
Indeed, compiling the firmware is more complicated than synthesizing a core. You basically just load the project file and hit 'go'.

Angel-child, 2020-01-17:
But you must not forget, you are geniuses and me only a user of your fantastic software who does this for the first time.  😉 The Problem still begins with the not working nstallation of quartus software.... 😂

Angel-child, 2020-01-18:
The Windows Version worked without Problems. And what can i say..... The VGA output with  15khz: 
[image]
It WORKS!!! 😁😁😁😁 FANTASTIC, GREAT JOB, THANK YOU SO MUCH!

closed 2020-01-18 by Angel-child

gyurco, 2020-01-18:
Cool. Added the setting to the Atari STe core, too.