mist-devel/mist-binaries #70: Drive sound with buzzer

Angel-child, 2020-02-02:
Hello again... I enjoy the 15khz change youve done so much! Now my next Question... Is there a way to implement a drive sound also with a buzzer upgrade. I dont find anything about that, so i think its not possible?
Best regards Torsten!

harbaum, 2020-02-02:
There is no buzzer in the mist. But you could hack together a little drive sound code and mix that into the regular audio stream.

Angel-child, 2020-02-02:
Ah ok i thought perhaps a buzzer is upgradeable in the mist on some pins internal...

gyurco, 2020-02-02:
You can connect one to the serial port, but a core should be modified to use it. And what core would benefit from a buzzer without LP filter and sigma-delta DAC? ZX maybe, or Apple II.

Angel-child, 2020-02-02:
Minimig-aga 😁 i meaned it only for the floppy drive sound... in the original minimig Board  it was implemented in core and a buzzer could be connected to two boardpins...So i thought its also possible for the mist

gyurco, 2020-02-02:
It can also be redirected to the UART pins...but it's not really interesting for me :)