mist-devel/mist-binaries #71: Vertical Arcade Games possibility to face the same direction ?

Gerloc, 2020-02-16:
Most Vertical Arcade game cores in Mist display seems to have a left rotation, But some have the opposite direction and no way of rotating the display around in mist. Like moon cresta having a right rotation.  So if you have a crt or flatscreen setup for vertical Arcade games its not practical to rotate the Whole crt or flatscreen display all the way around for just some of the games. So is it possible to have all the vertical Arcade game cores facing the same direction as a rule ? , Most likely left since most of those already faces this way. I don't know why Moon Cresta and some other games have a right rotation ? . Maybe it was this way originally..?

sorgelig, 2020-02-16:
FPGA recreates original arcades as much as possible. There was no standard of rotation, so each arcade cabinet used its own design. That's why some arcades rotated CW some CCW, there are even arcades rotated at 180 degree being a normal horizontal arcade. Just because of HW engineer decided to turn the monitor in his arcade.
So the way different arcade cores displayed is exactly how it was in original HW. And unlike in software emulators you cannot simply rotate to any direction. Video is constructed from live signals, not from the buffer. 
So, to rotate it, FPGA needs to use additional frame buffer which will be rotated later.
Some (few) arcades can be re-designed to shuffle the sprites/tiles coordinates - this is not a trivial task too.
Simple wish - hard realization.

Gerloc, 2020-02-16:
Ok. Thanks for the Answer, At least Nice to know how this is. Well Rotating a flat screen around is possible, But for a Crt screen its not so easy. Exept perhaps buling a Coktail Arcade, That would work. Anyway Nice to see so many Arcade game cores in mist and the other Fabolous cores. Thank you.

Gerloc, 2020-02-19:
So about Cocktail versions, I see that Moon Cresta had a Coktail version, Doesn't Cocktail Versions have the ability to flip the screen in hardware itself ? Then in would be a good idea to add Cocktail Versions of the Vertical Arcade games that lets say rotates to the right, since most of the others seem to rotate to the left, then by flipping the Cocktail Versions in the hardware , Then many more of the vertical Arcade games could face the same direction..  Should be possible ? if it is, this seems to be a good idea.  The by adding the flipscreen flag to the menu should be enough to flip the screen in cocktail versions..?

sorgelig, 2020-02-19:
flipped mode in cocktail is surely HW assisted, but it still controlled by software which is aware of current rotation and may implement some rotation in software as well. So it's not automatic fix. It needs to be investigated for each arcade for possibility of such workaround.
Hobby projects like MiST is driven by self-motivated developers. So usually development is started from something which needs to be fixed but no one wants to do it for different reason. So you can become such self-motivated developer. Otherwise your wish may won't be fulfilled ;)

Gerloc, 2020-02-20:
I Would like to be a Developer for mist if i had the knowledge to do so, But my Programming skills are not up where it need to be to do so, But i get Your point.  Solutions tends to pop up if the Developer community seems it fit to do so, We users can only give our meanings and somtimes test and bug reports and ideas that sometimes could be useful.

gyurco, 2020-02-20:
Some cores actually implement coctail mode with flipping, but enabling it actually mirrors the screen for the subsequent player. I don't think it helps to have left rotation for player 1, and right rotation for player 2.

sorgelig, 2020-02-20:
Cocktail mode doesn't flip the screen, but rotates at 180 degree. 
Still HW rotate abstraction from SW is varying from arcade to arcade.

Gerloc, 2020-02-20:
Yes player 1 and 2 would have the screen flipped around, But maybe having the flip flag enabled in menu mean that by using the menu one could flip the screen around on the fly for player 2, Cumbersome yes , But still more practical than rotating the monitor around, specially for crt screens, Or maybe have sort of cheat that enabled player 1 and 2 in the direction of ones choosing, But that would probably be like hacking and maybe a Place where mist cores should not be ?

closed 2020-02-20 by Gerloc

jotego, 2020-02-21:
I have added flip support on the menu for **1943**, which doesn't support it originally. The game board actually has a flip signal. I guess that it is used for cocktail mode but I haven't tried cocktail mode, actually. When I let that signal toggle at the user will you get the effect that @Gerloc is looking for but as @sorgelig pointed out, as software is not ready for that you do get a row of background with a wrong colour on the title screen.
There is actually some headache associated with the flip option, because once you offer it you have to be sure it works correctly so it is not as easy as just bringing the option to the menu.

gyurco, 2020-02-21:
And if you implement it in one game, then the other 99 will still won't have it. It has to be done one-by-one.

Gerloc, 2020-02-22:
Yes. So then this issue seems to be clarified here.  Thanks for all the answers.. Maybe parts of this info could be in the docs, About the vertical Arcade games can have opposing direction in mist and the reasons for it.