mist-devel/mist-binaries #72: Amstrad CPC464 Mode in Amstrad Alt.

Julitium, 2020-03-24:
Hi.
How can I use the core in CPC464 mode?
It is awkward for me to use the | tape command every time I reboot the core.
Thanks.

gyurco, 2020-03-25:
I think you need a new romset for 464.

Julitium, 2020-03-25:
If you refered to this.... 
Boot ROM has following structure:
OS6128 + BASIC1.1 + AMSDOS + MF2 + OS664 + BASIC664 + AMSDOS + MF2
Not working (Tested)..... Because CPC464 NOT use AMSDOS. 
The solution is to create a third mode without AMSDOS. 
I have a new problem to create a Amstrad.rom, MF2. Rom have 8kb, Not 16kb. But this Not very important.

Julitium, 2020-03-25:
Sorry, i make a Amstrad.rom without AMSDOS rom.......
Its Works for me. 
Very Thanks for Help me.

closed 2020-03-25 by Julitium