mist-devel/mist-binaries #73: Glitches using AGA chipset.

veb-xenon, 2020-03-31:
I've been updating some games and testing them in whdload (I download the Whdload games from EAB File Server) . Now when I try to execute Street Fighter II (yes, I know there are better versions, bla bla bla), game runs, but there are some graphical glitches that there weren't in previous cores. 

ftp://ftp:amiga@grandis.nu/Commodore_Amiga/Retroplay/Commodore_Amiga_-_WHDLoad_-_Games/S/StreetFighter2_v1.4_0882.lha

On Whdload works perfect. No glitches. I'm using lsato core on Mist, 68020, 24 MB RAM and no turbo (on AGA games only I use the turbo function). Last Whdload and Amiga 3.1.4. 

Fixing this problem could help fixing other glitches... The problem it's AGA related. If I change the chipset to ECS game works, so something on AGA is not working well now on last updates.

veb-xenon, 2020-06-15:
I've updated since the core, and these glitches are still there using AGA on the same game.

Torti-the-Smurf, 2024-01-13:
The JST version works flawless. (if you really want to play this .....)  ;)