mist-devel/mist-binaries #74: C64 Graphics Glitches

Julitium, 2020-04-01:
Hi.
The latest C64 core (C64_200401) still has graphic glitches.
It has been greatly improved with the VIC II modifications.
Test:
Trick and Thread (DEMO).

Thank you very much for the improvements.

gyurco, 2020-04-01:
You meant Trick and Treat?
That demo is 37 minutes...even if you would be specific where's an issue in it, I won't debug that. If some future updates will fix it, then good.

Julitium, 2020-04-01:
Hi.
This happens on the first disc .... No more than 5 minutes, after the horizontal scroll.
Although you do not repair it, thank you very much for the improvement made. This core works great.
Greetings.

Julitium, 2020-04-01:
Hi.
The most curious thing is that before the improvements this part worked well, but the horizontal scroll was a complete glitch.
Again, thanks so much for the improvement, it's great.
I also think that it is very close to being perfect.
Greetings.

gyurco, 2020-04-02:
The only glitch I noticed on the first disk is some minor flickering of the upper and lower border area of that long vertical twister.

Julitium, 2020-04-02:
Hi. 
Right, I'm the only glitches I've noticed, too.
Taking into account that it is a demo that tests the operation of the VIC-II at 100%. The results are great.
Again, thank you very much for the improvements to the VIC-II.
The exact Cycle is very very close.
Greetings.

raparici, 2020-04-05:
Hi Gyurco,

VIC-II has so much been improved ! Thank you. However a vertical glitch appeared in many demos. It follows the left Vertical border. This is not showing in previous cores. I made an album of before and after that you can see here:

https://photos.app.goo.gl/gHoNoiz5Poorn8sy7

I put the demo names inside the pictures. These are though:

Deus ex Machina 
Edge of Disgrace 
Performers Cebit 2018

This is the only issue I came across after many complex demos. Your VIC-II is almost perfect !

lroby74, 2020-04-05:
> Hi Gyurco,
>
> VIC-II has so much been improved ! Thank you. However a vertical glitch appeared in many demos. It follows the left Vertical border. This is not showing in previous cores. I made an [...]

I saw same issues in Edge of Disgrace too, with previous C64 core, these are not present

raparici, 2020-04-05:
The previous issues are definitely fixed. This new VIC-II version is a heck of a lot better but here I am talking about a glitch that was introduced in the recent version. Do you agree?

Julitium, 2020-04-06:
Hi.
I do not entirely agree. In previous versions the failures were more numerous and larger (some a complete disaster). In the new version they were reduced to the minimum expression (some problems related to the border).
The new version is so much better.
Thanks again Gyurco for this huge step.
Greetings.

gyurco, 2020-04-06:
Fixing the remaining issues are not trivial, since it's not enough to have cycle-exactness in the CPU clock level (1 MHz), but must be cycle-exact in the pixel clock level (8 MHz). It means 8 times more possibilities for each register change to manifest its effect.

sorgelig, 2020-04-06:
if cpu clock will be bound to specific pixel phase of 8MHz, will it fix the issue?

Julitium, 2020-04-06:
Hi.
I think those 8Mhz are incorrect. Please check these pages for more accurate information ...

https://codebase64.org/doku.php?id=base:cpu_clocking

Perhaps this can resolve this matter.
Thank you so much.
Greetings.

sorgelig, 2020-04-06:
Which part is not correct? The master clock absolute value doesn't matter.

Julitium, 2020-04-06:
Hi.

"PAL C64 master clock: 17.734475 MHz

NTSC C64 master clock: 14.31818 MHz

The CPU frequency is then calculated from that by simply dividing the frequency by 18 (PAL) or 14 (NTSC). The VIC-II runs at a frequency which is exactly 8 times that of the CPU. This is the so called “dot clock” which has to be very precise in order to keep the right timing needed to generate a video signal compatible with all TVs. The CPU of the time could not go that fast, max. 1MHz, but the CPU still needs to be phase synchronous to the VIC-II because they share control of the address / data bus of the machine. That's why the VIC-II internally provides a clock divider which feeds the CPU.

CLOCK_PAL = 985248 Hz
CLOCK_NTSC = 1022727 Hz

CLOCK_VICII_PAL = 7881984 Hz
CLOCK_VICII_NTSC = 8181816 Hz"

So it is not 8Mhz if not, 7.88Mhz.
Thank you.
Greetings.

gyurco, 2020-04-06:
@Julitium thanks for linking a hardware basics page :dagger: Maybe I know what I speak about...It's not important if 7.88 or 8, but the relation of the CPU and the VIC-II clock. Simply there are 8 times more possibilities. Or 4 times, if you assume that the CPU writes/VIC-II latches at the PHI0 positive phase. However there are internal delays inside the chip, too.

@sorgelig it's currently linked to one specific phase (pulseiord and pulseiowr signals). Mostly good, but there are some sprites tests (and issues with sprites) which probably need revisiting the register latching times.

Julitium, 2020-04-06:
Hi.
I understand. I am sorry.
😔
Thanks again.
Greetings.

gyurco, 2020-04-06:
No problem.

gyurco, 2020-04-07:
I think I fixed all these regressions, like the vertical bar garbage, and some others in the Chorus Rocketry demo, but I didn't want to retest all the demos for other regressions. So please try it, and report if something broke what worked previously:
https://github.com/gyurco/mist-board/releases/download/xx/C64_mist_test.zip

sorgelig, 2020-04-07:
I didn't test MiST version, but if that binary is compiled from latest public C64 source, then nothing is fixed. I've copied all changes to MiSTer and tested yesterday.
But it's possible i've missed something.

sorgelig, 2020-04-07:
oh, i see the source is attached release is quite different. Is it reverting of recent changes?

gyurco, 2020-04-07:
Source is there:
https://github.com/gyurco/mist-board/tree/c64-2
I will put in the main repo if it works.

sorgelig, 2020-04-07:
i've tested the changes on MiSTer:
DeusEx no problem. 
Edge Of Disgrace almost ok. Some vertical line briefly appears at the near left edge.

Julitium, 2020-04-07:
Hi.
The new version has the same problems in "Trick and Treat".
In "Edge of Disgrace" it has improved slightly.
I keep testing.
Note: if you switch to NTSC mode the problems sometimes disappear and other times they get worse.

Thank you very much Gyurco for the new version.
Greetings.

sorgelig, 2020-04-07:
C=Bit18 is fixed too.

gyurco, 2020-04-07:
NTSC mode is not tested by me at all. Also I think most demos are written for PAL machines. 

The Trick and Treat border issue is more likely CPU problem (the border opening switch sometimes slips 4-5 cycles into the next line, where it's ineffective). Maybe INT is still delayed too much (4-5 cycles can be a single long instruction). Would be good to have a PRG file which starts with that part. Then it can be debugged faster.

Will look at EoD.

gyurco, 2020-04-07:
The good news is EoD glitch caused by sprites, not the background layer (the sprite is positioned to the place when the shift register reload should happen). This is another big topic to play with.

Julitium, 2020-04-07:
Hi.
Bad news ..... Certain tape loaders that DO NOT work.

Cyberdyne Warrior and Cybernoid 2 for example.

Result is System Reset.

Under these circumstances, this is a step backwards in my opinion, please check this, I do not understand the relationship.
I go back to your latest official version.

Thanks.
Greetings.

gyurco, 2020-04-07:
This one should behave better:
https://github.com/gyurco/mist-board/releases/download/test2/C64_mist_test2.zip

Julitium, 2020-04-07:
Hi.
The affected loaders came back to life. Brilliant.
I will try this better and try the demos again.
Thank you very much Gyurco

Greetings.

sorgelig, 2020-04-07:
I still see several times briefly blue vertical line in EoD near the left edge of screen in that checker field.

Julitium, 2020-04-07:
Hi.
Well, what I see is an improvement over the improvement.
The previous functionalities seem intact.
Brilliant.
The exact cycle is even closer.
I keep testing.
Again Thank you very much Gyurco.

Greetings.

gyurco, 2020-04-08:
Sprites are still not fully OK (the brief blue line). However every attempt to fix it caused breakage in Krestage 3. Also 3rd screen of Errata from Emulamer is not improved.

Julitium, 2020-04-08:
Hi.
After testing quite a few demos and previous functionalities, I think the effort has been a success.
The biggest problems have been a line at the edge of the screen.
Thank you very much Gyurco .... I continue testing.

Greetings.

Julitium, 2020-04-08:
Hi.
Testing tapes I have noticed that the Ocean New Tape 4 loader does not work (it never did). Examples:
Lethal Weapon
Outrun Europe
This is sensitive to the precision of cycle numbers.
Perhaps it is a clue to the correct functioning of demos
Thank you.
Greetings.

Julitium, 2020-04-10:
[image]
[image]

gyurco, 2020-04-10:
You're knocking on already opened doors. If the whole thing would be as simple as using the published datasheet, then every core would be already perfect.

Julitium, 2020-04-10:
Hi.
OK.
Perhaps it is a communication problem, previously I seemed to understand that the delays of the chip were not known. After that, the problem was the CPU VS VIC-II clock ratio .... In short, I'm sorry. Excuse me. I will dedicate myself to the beta test.
Thanks for the improvements.
Greetings.

gyurco, 2020-04-10:
Yes, delays are not known - inside the chip. These AC characteristics is for cooperating with external parts, like the CPU. However they're not really relevant for us, as all things are in the FPGA.
But here's an another test version, hopefully with improved sprite handling (at least it passes a lot of VICE tests now):
https://github.com/gyurco/mist-board/releases/download/test3/C64_mist_test3.zip

Julitium, 2020-04-10:
Hi.
I understand.
I am going to try this new version Now... Loads, demos, etc.
Thanks for the Gyurco improvements.
Greetings.

sorgelig, 2020-04-11:
EoD is fixed

gyurco, 2020-04-11:
Good. I just cannot figure how the shift register should behave around the reload point, that's the Achilles' heel of the Emulamer demos.

raparici, 2020-04-11:
WOW! what a great core ! good job Gyurco and everybody assisting this. This is the first time I feel we have an open core that compatibility wise is on par with commercial ones (TC64, U64). The list of any pending issue must be really minimal. @Julitium can you provide the list on what is yet pending that you know of? My list is totally clear. Maybe we can freeze demos and provide Gyurco with PRGs that go straight to those issues.

cheers everybody

gyurco, 2020-04-11:
Can I request a PRG from the Trick and Treat demo at this position?

https://www.youtube.com/watch?v=SWy5CBUvr0E&feature=youtu.be&t=987

raparici, 2020-04-11:
@gyurco let me know if this works for you:

https://mega.nz/file/DR1xhQwS#ppS7BUEcauNN0fI1D4ysjfTtYKw7wvYUw4oHUJkwvVY

Julitium, 2020-04-11:
Hi.
I have been testing all the material that I could .....

Trick and Treat (same fails ... Skulls could be a VSP bug aka delay DMA bug ??).

EoD = Simply perfect !!

Krestage 1 and 3 = not detected as emulation .... Runs perfectly.

Eat Light 13 = Perfect.

C: 18bit = perfect.

Emulamer demos ...

Conclusion = 50% not scroll text but it is not detected as emulation. Well.

RAM bo = OK.

Ruined Art = Fail (Clue is non-visibility sprites related)

SONG25 = OK

Undead = Fail (no scroll vert.)

Bad-copy = Fail (Clue is Illegal opcodes related)

Emulamer advises to take a look at texts ...
Marko makela (VIC 2-PAL)
Christian Bauer (VIC-II Application in C64)

Games loading ok ....
Except Games with Ocean New Loader 4 (Tape)

It is also some other flaw in both Tape and D64.

The improvements have been beyond imagining, thanks Gyurco.
If you need something...
Beta test ... PRG .... Or whatever, you have nothing else to ask for.
Thanks again.
Greetings.

gyurco, 2020-04-12:
@raparici Perfect! And for an added bonus, I could listen to Alexander the Great theme without stopping.

raparici, 2020-04-12:
Good.

This is the freeze of Trick and Treat at the vertical 3d ticker bar that Julitium commented about. This time I couldnt make a single PRG to work and you need to load"*",8 and run and wait till it loads the memory dump from the disk attached below. Not as fast as the prg but still more convenient than going through the demo.

https://mega.nz/file/jJMDhS7L#qJg0WpGoUnGe4HMVNao_waZbsyKnaB-6OveGuLIC6YQ

gyurco, 2020-04-12:
Here's with a small CPU change for a perfect Trick and Treat:
https://github.com/gyurco/mist-board/releases/download/test4/C64_mist_test4.zip
As I suspected the IRQ latency was still too big in some cases.

Julitium, 2020-04-12:
Hi.
I test everything again.
Gyurco, congratulations. You are a master.
I will report what I find, but I think it will be little or nothing.
Thank you very much for the Gyurco improvements.
Greetings.

lroby74, 2020-04-13:
I've tested this C64 core with whole Edge of Disgrace demo, and it's perfect, any issues present!
I love you! Thanks a lot!

Julitium, 2020-04-27:
Hi.
All the demos and games I've tried work satisfactorily except for some emulamer demos.
The tape games that have the "Loader Ocean New 4" still do not work, but I suppose it is a Handler Tape problem.
For me the improvements are very important and represent a before and after.
Thank you very much for all the improvements Gyurco.

do you want to close the issue?

Are we still testing the version?

gyurco, 2020-04-28:
Emulamers are work in progress, there won't be more test versions I think.

gyurco, 2020-04-30:
The new release (200430) should run Emulamers. Only one 6569 glitch is not emulated, but that's so stupid, I skip that.

raparici, 2020-04-30:
haha @gyurco you did it again. I can say the private cores like U64 do not run the emulamer undead demo well while this open core does. So many congrats and thanks for your contribution to the community and to help the hardware description of these lovely machines to transcend over time.

raparici, 2020-04-30:
Hi @gyurco I run a bunch of demos and found an issue. Now in DeusExMachina the demo fails to continue from the second disk side. It crashes. All the rest is fine.

raparici, 2020-04-30:
mmmm maybe it was Jiffy... I'll try again now.

Julitium, 2020-04-30:
Hi.
@raparici you have tried to reinsert the disk again? (I noticed the same thing before) but reinserting the disk several times continued, I did not give it much importance.
Regards.

Julitium, 2020-04-30:
@raparici, I tried Deux, this works fine (with standard ROM) disk change works. It may be the Jiffy's problem.
Regards.

Julitium, 2020-04-30:
Hi.
@gyurco , when I thought you couldn't improve your progress, you go and improve the improvements.
Brilliant.
I am trying everything again and at the moment it is a resounding success.
You are certainly a genius.
I still testing. 
Many thanks.

closed 2021-07-15 by gyurco