mist-devel/mist-binaries #76: Analogue sticks

jotego, 2020-04-03:
I am trying to add support for analogue sticks to use in the[ Forgotten Worlds ](https://www.youtube.com/watch?v=PYJfoesXYC8)arcade. I was thinking of moving the character using the d-pad and then rotating him with the analogue stick. But I see that MiST actually links both the d-pad and the analogue stick so I receive inputs for both at the same time. I tried connecting a dual analogue stick game pad but the second stick doesn't seem to be read.

On top of that, Forgotten Worlds is a two player game, so I actually need to do the same for the second character.

I can edit the user_io module provided the firmware already supports this. But does it? And how?

Thank you

gyurco, 2020-04-03:
The firmware sends the digital joystick events as analogue ones, too (mapping to min-max values). If I understand you correctly, this is unwanted?

jotego, 2020-04-03:
Yes, even if the left analog stick and the d-pad are linked, that's fine if I can use the right analog stick.

gyurco, 2020-04-07:
Well, that would need checking the USB part, but I don't have such controller to test with. I suspect the two parts are multiplexed somehow, but I don't know how it's done.

closed 2022-01-17 by gyurco