mist-devel/mist-binaries #79: Minimig mouse acceleration and glitching issues

famiclone6502, 2020-06-26:
The mouse movement in the Minimig core is currently very sensitive with my wireless laser mouse. I am not necessarily requesting sensitivity adjustment as a new feature, but as a bug fix: if input is "too fast" it should cap maximum speed the Amiga mouse should ever accelerate.

If I move "too fast" it will glitch and throw the cursor all over the screen. It does not seem to break games, but I have to be careful and slow with my movement. I only need to move very slowly for the cursor to travel a large distance. 

A good example would be when running Lemmings or even just Workbench.

kowoba, 2020-06-27:
And what are your pointer settings in Prefs/Input?

robinsonb5, 2020-06-27:
Prefs/Input settings aren't relevant if the mouse counters are overflowing before the Amiga reads them.  Nor are they applicable to games such as Lemmings.

gyurco, 2020-06-27:
If you can compile, please try with this patch:
https://github.com/mist-devel/minimig-mist/commit/d7c8ea74ff4af551a0d3ca8c41ee6374dfd7fa97

famiclone6502, 2020-06-27:
Sorry, I cannot compile at this time. 

I've confirmed that the previous release minimig_mist_20200209.rbf did not have the issue.

gyurco, 2020-06-29:
@robinsonb5 Good to see you here! I've added the mouse wheel driver disk to the binary repo, which I took from your TC port. However it crashes in 68000/ECS mode, and works only in 68020/AGA. Is it compiled for 68020 only, or I'm doing something wrong?
This is the ADF I talk about:
https://github.com/mist-devel/mist-binaries/blob/master/cores/minimig-aga/WheelDriver.adf

robinsonb5, 2020-06-29:
It's probably built for 68020 - I remember some years back being asked for a 68000 build of FreeWheel, and since the WheelDriver's an evolution of a project dating from the same time it probably needs the same build tweaks.  I'll take a look later on.

(Most of what appears in my dev branch in the coming weeks should be applicable to MiST as well as the TC64, by the way - I'm trying not to break other targets too badly or make changes that are too invasive, but it will no doubt need some adapting.)

gyurco, 2020-08-12:
I think this bug was fixed in the latest release.

famiclone6502, 2020-08-12:
Confirmed fixed. Thank you everyone!

closed 2020-08-12 by famiclone6502