mist-devel/mist-binaries #82: BBC micro, enable tape loading

rcmolina, 2020-07-12:
Thanks for your work. 
Any option to enable loadin from tape as in other cores: Spectrum, C64, so this core re-creates real the reak machine funcionality?

Thanks in advance.

Rafael

gyurco, 2020-09-23:
It's a custom chip hooked on a 6850 ACIA which handles the tape input. One has to be very motivated to implement it.

rcmolina, 2020-09-23:
I know what you mean, but this is not a money question, just getting the schematics from rhe okd company or reversecengineering it. I was very surprised user group people did not push it.. Sooner or later old hw will not survive. I contacted also with hoglet, may be the future will smilevus with the missed schema of the ferranti serial ula. Hope to be alive when that happens.

gyurco, 2020-09-23:
It's a custom chip, you won't find any schematics for it (unless somebody decaps it, and analyze the photos). However it's not needed, the functionality is probably known, just it's not a trivial thing like on Speccy, where you connect one pin to another.
But anybody who's interested can work on it, it's not forbidden.

harbaum, 2020-09-23:
I don't think anyone is asking for money, here.

Main motivation to implement these things usually is to be happy and proud of the result and to achieve some kind of personal goal like e.g. 'being able to this old game perfectly that most emulators fail to'. 

For someone who does not see a personal advantage in a certain technical aspect it may be hard to find the motivation to spend the hours needed to implement something like that.

I once implemented the c16 core without any interest in the device itself. This was more a psychological endeavour than a technical one 😁

rcmolina, 2020-09-23:
@harbaum , I  agree, sorry I didn't explained myself very well. David (hogglet) explained me: "The BBC Serial ULA is more of an unknown quantity. Very little is known about how the tape data seperator actually worked. There are no original schematics, and I know of no one that's decapped one for reverse engineering".
There are three parts to the tape interface on the Beeb:

the analog signal conditioning (3 op amps, that amplifiy, filter and square the signal)
the serial ULA (this does frequency detection and clock/data separation)
a 6850 ACIA (converts synchronous bits back to bytes)
(2) and (3) would be straightforward to implememt in the FPGA

(1) is the tricky part. You either need to replicate the analog circuitry from the Beeb, or you need an ADC and do this digitially.

I was asking him cause I couldn' urdestand why Acorn Electron tape implementation could not be used here.

Anyway, if you need such a motivation, it should be very tricky and time wasting :(