mist-devel/mist-binaries #83: Real ColecoVision controller unsupported on MiST DB9 port in Coleco core

famiclone6502, 2020-07-14:
Only the directions and one of the two action buttons are supported for plugging up a real ColecoVision controller. None of the number pad keys are recognized. I did not test these, but additionally the Rollerball and Super Action Controller (spinner and 2 additional action buttons) functionality are assumed to be currently unsupported in the ColecoVision core. 

[Here](http://www.chromesphere.net/coleco/Controller_Interface.html) is some technical information about the controller that may help.

I was able to use the Coleco core to repair a directional pad using a controller test program, but I was not able to see if I fixed the secondary button or number pad.

gyurco, 2020-07-14:
Of course it's not supported. The pinout differs from the Atari DB9 joystick.

famiclone6502, 2020-07-14:
I apologize, I did not realize it was physically impossible to implement. I wasn't sure how it was connected to the FPGA under the hood. 

Closing.

gyurco, 2020-07-14:
It's not connected to the FPGA at all, it's connected to the ARM MCU.

famiclone6502, 2020-07-14:
I'm interpreting that it isn't technically possible.

gyurco, 2020-07-14:
I don't know, one have to investigate the connections and change the firmware accordingly.

famiclone6502, 2020-07-14:
If it is technically possible, I will re-open.

gyurco, 2020-07-15:
As pin 8 is NC, I don't think it's possible.

robinsonb5, 2020-07-15:
Worse still, pin 7 is common, which means any direction or button on a Colecovision controller will short the MiST's +5v line to ground.  You're going to need a wiring adapter, otherwise the magic smoke is going to escape.

gyurco, 2020-07-15:
I found this:
https://i1.wp.com/www.retronicdesign.com/wp-content/uploads/2014/11/ColecoController.jpg
I don't see pin 7 connected, so it might be safe. Just won't work.

famiclone6502, 2020-07-15:
That one is a good reference, it's the same as [this](http://atarihq.com/danb/files/ColecoController.pdf). If pin 7 isn't physically connected on MiST and it's used by the Coleco controller, then it won't work. 

Thanks for investigating.

closed 2020-07-16 by famiclone6502