mist-devel/mist-binaries #85: Savestate on Commodore 64

fiorefr73, 2020-08-04:
First of all thanks for the fantastic job! I really enjoy the Mist cores.

Just one question: how difficult would be to create a save state/load state option in the Commodore 64 core?
Something like the 'Save Snapshot file'/'Load Snapshot file' functionality in VICE.

In theory it should 'only' write the content of the RAM into a file in the SD to save the state and write back the content of the file in RAM to restore it. Am I making it too much simple?

Thanks in advance for any reply.
Francesco

gyurco, 2020-08-04:
_In theory it should 'only' write the content of the RAM into a file in the SD_ 
and the internal state of all chips, RAM of the 1541, etc...
Use an Action Replay cart instead.

jotego, 2020-08-05:
You need to keep the state of all flip flops of the CPU even. It really is a lot of work.

fiorefr73, 2020-08-06:
> _In theory it should 'only' write the content of the RAM into a file in the SD_
> and the internal state of all chips, RAM of the 1541, etc...
> Use an Action Replay cart instead.

Thanks for the reply.
I will investigate how to use the Action Replay cart in the FPGA

fiorefr73, 2020-08-06:
> 
> 
> You need to keep the state of all flip flops in the CPU even. It really is a lot of work.

Thanks for the answer.

closed 2020-12-02 by gyurco