mist-devel/mist-binaries #86: Retrobit Pad mapping for fpgagen

sprudel79, 2020-08-12:
Hi everyone,
thanks for your awesome work, all the cores for the MIST are just amazing.
I would like to use my new "Retrobit SEGA Mega Drive 8-Button Arcade Pad - 2.4 GHz Wireless" controller with the fpgagen core. As I was able to read in the MIST docs the mist.ini needs to be adjusted.
The current mapping looks like this:
joystick_remap=0F0D,00C1,1,2,4,8,200,20,10,100,400,800,1000,2000,1400,80
(Forget the ninth number because I was trying a few things there)
My reference game for testing is Blockout, since it makes use of the A+B+C+Start button.
With the config above I am able to use A+B+Start without any problems.
Now I am stuck with the C button because I cannot figure out which button to map there, I don't know which button the core actually requires. Using the Joystick test on the MIST I can see that X+Y buttons are also mapped but using them doesn't cause the right result.
Has anyone else achieved to do the entire mapping for the Retrobit controller?
Thank you very much in advance.

sprudel79, 2020-08-12:
I've figured it out by myself...the core is using the select button as the C button on the pad.
I ended up with the following mapping:
joystick_remap=0F0D,00C1,1,2,4,8,200,20,10,100,400,800,800,40,40,80 
C button is button no 8, so I've kept the select button to be also select. Besides I've mapped Z as R, maybe I will improve this soon to support also 6 button games.

Julitium, 2020-12-07:
Hi.
You must try Mapper tool for PC.... Useful really. 

Regards.

boomlinde, 2021-04-07:
I got the same pad today, except wired. IDs seem to be the same. Here's a 6-button configuration:

    joystick_remap=0F0D,00C1,1,2,4,8,200,20,10,100,400,800,400,40,800,80

Some of these entries aren't necessary but it works now so I won't touch it!

Mapping was a right headache, even with the mapper tool, unfortunately. Windows 10 seemingly used a driver for another device which switched some button pairs around and gave the wrong IDs. Then there's fpgagen's internal mapping of MiST buttons to MD buttons. For example, what MiST calls Select is C in fpgagen.

boomlinde, 2021-10-04:
Here is a Python script I've made to simplify mapping: https://gist.github.com/boomlinde/3108517639b70d772275f8499c908ff5

It includes a mapping definition for this controller for some cores and arcade games. Eco Fighter is quite comfortable with this controller, using the shoulder pads for aiming :)

In short, the joystick_remap config maps gamepad buttons to a virtual MiST controller, which resembles a SNES gamepad with four face buttons, start, select and shoulder buttons (with two extra pairs of shoulder buttons). The cores then individually decide what buttons on the virtual controller correspond to buttons for that core. For some cores that are basically subsets of the SNES controller this is done in an obvious way. The SNES and NES cores for example use the virtual controller buttons 1:1; A is A, B is B, etc.

My own troubles with mapping seemed to be due to a misunderstanding of the "direction" of the mapping. The indices correspond to USB joypad buttons, and the numbers to the internal MiST controller. The Retrobit Genesis pad with shoulder buttons has the following order of physical buttons:

right, left, down, up, y, b, a, x, l, r, z, c, mode, start

In the joystick_remap config field, you'll need to list numbers corresponding to the virtual MiST controller buttons in this order.

The confusion is compounded by the fact that the mapping of buttons to the fpgagen core from the virtual MiST controller isn't obvious. Directions, A, B, X, Y and Start correspond directly to the MiST controller buttons. The rest are different:

* `mode` in fpgagen uses `r` on the virtual controller
* `z` in fpgagen uses `l` on the virtual controller
* `c` in fpgagen uses `select` on the virtual controller

gyurco, 2021-10-05:
Thanks for the efforts. The MiST internal naming is following the Nintendo layout.