mist-devel/mist-binaries #88: Sound very low

AxelStone, 2020-09-01:
Hello, it seems that the lastest version of the framework has decreased the volume a lot, the cores sound really low. To check this, compare the lastest core of CPC not implementing PlayCity with the first one implementing it (and using the new framework). The volume difference is very high.

With such low volume you need to turn up the TV volume a lot and get a lot of electrical noise.

Thanks so much.

gyurco, 2020-09-02:
Are you referring to MiST? The DAC module and the PSG volume didn't change with the PlayCity addon.

AxelStone, 2020-09-02:
Hello gyurco, yes the Mist core of CPC. I'm pretty sure that you didn't change volume in the core, this is why I mention that probably it's something with a new framework and default values. I can record audios comparing volume of both cores or even you can test easily, comparing the lastest one before PlayCity and the first one using it. Did they where compiled using different frameworks perhaps?

Thanks so much.

gyurco, 2020-09-08:
Did you set the output to Stereo? Mono output is louder. The only relevant thing changed is the PSG implementation, now it's the same as in the Atari ST. But that was before the Playcity.

AxelStone, 2020-09-08:
Yes, I get a bit louder sound but not comparable to previous core. I've asked other friends just to be sure that it's not only me and they have apreciated same effect. It's not critical (for example MSX core has much lower volume, you need to put TV volume very high) but it seems like cores tend to sound lower than previous versions. As I said, do a very easy test: try core before PlayCity and then the one with PlayCity, you will notice a clear difference. 

About the change you mention in PSG implementation, perhaps it's the cause. I'm not sure if my old core is the last one before PlayCity, I can check if you need.

Thanks. Best regards.