mist-devel/mist-binaries #89: Feature request: Soft reset for the C64

philipp685, 2020-09-23:
To make the C64 core even more awesome (thanks for all the great work on it, I love it!), I'd like to have a soft reset feature that doesn't initialize the RAM. Context: When programming, it would be useful to be able to trigger a reset without losing everything in memory.

Basically, a reset option that doesn't [set force_reset](https://github.com/mist-devel/c64/blob/master/rtl/mist/c64_mist.vhd#L933), be it a new menu entry or a key combo.

gyurco, 2020-09-25:
I can make the reset key (ctrl+F11) act as a soft reset.

philipp685, 2020-09-25:
That would be great!

gyurco, 2020-09-25:
Try the latest release!

philipp685, 2020-09-25:
Wow, that was super quick! Works like a charm, thank you so much!

closed 2020-09-25 by philipp685