mist-devel/mist-binaries #90: Amiga RTG error. Monitor not support 31KHz / 49Hz

JorsuSHerluN, 2020-09-24:
My monitor not support 31KHz / 49Hz configurarion of core "minimig_mist_rtg_200923.rbf"
How can I change the Mist settings to 31KHz / 60Hz?

gyurco, 2020-09-24:
For RTG modes? A Picasso96Mode utility is installed in Prefs with the Picasso driver, it can be used to create custom modes.

JorsuSHerluN, 2020-09-24:
At the start of the core. I can't see neither the startup screen showing cpu type, memory and the ball.

gyurco, 2020-09-24:
Did it work with the non-rtg core? It was 50 Hz also, the core is mostly PAL.

JorsuSHerluN, 2020-09-24:
All previous AGA core work perfectly for me. The problem is only in RTG cores.

gyurco, 2020-09-24:
Something was changed around the sync signals. Maybe @robinsonb5 has an idea.

robinsonb5, 2020-09-24:
I'm at work right now - will take a look later.  Is there a way to make the core start up in NTSC on MiST?  (An INI option, perhaps?  On other platforms you can hold an F key at boot.)

gyurco, 2020-09-24:
I didn't know until now - you can use F1 for NTSC and F2 for PAL:
https://github.com/mist-devel/mist-firmware/blob/master/config.c#L376

JorsuSHerluN, 2020-09-24:
The AGA core works fine for me at 31KHz / 50Hz. The RTG core works badly for me at 31KHz / 49Hz.
Is it possible to change the RTG core to 50HZ?

robinsonb5, 2020-09-24:
Ah, I think I know what this is.  The MinimigAGA codebase uses an NTSC base clock throughout (~7.15MHz) even though it defaults to PAL - this means the framerate is slightly higher than it should be when in PAL mode, and audio plays back slightly sharp.  My codebase uses a PAL base clock (~7.09MHz) which means the framerate is "correct" but being slightly lower its upsetting your monitor.  Should be a trivial fix.  (In the long term it would be nice to use a reconfigurable PLL to fix this.)

gyurco, 2020-09-24:
Then that was the reason of the reported jerky scrolling with the old clock on some monitors, so reverting it won't be optimal.

robinsonb5, 2020-09-24:
Oh yes, I hadn't made that connection, but I think you're right.  Using the PAL base clock isn't correct when the core's in NTSC mode though - framerate will be slightly too low, and sounds will be slightly flat.  I'm thinking the best solution might be ALT_PLL_RECONFIG and an option in the INI file?

gyurco, 2020-09-24:
Yes, the reconfigurable PLL is a good ide, but why need the ini file? The NTSC/PAL switch in the OSD (or F key during boot) is not enough? Only need to expose the setting to the top-level.

JorsuSHerluN, 2020-09-24:
When I change with the OSD button the frequency is 15KHz/50Hz. It also does not work on my monitor.

gyurco, 2020-09-24:
VGA monitors seldom work in 15kHz. It's for TV sets (or a C1084).

robinsonb5, 2020-09-24:
We need the INI file setting because in rare cases like this one, it's necessary to use the machine configured for PAL but using the NTSC clock setting.  (i.e. how the core used to be)

gyurco, 2020-09-24:
I would expect the NTSC/PAL switch on the OSD act as a master switch, and it can be different in each profile . Maybe the ini can be used for override only (e.g. force_crystal=none/ntsc/pal with none as default)

robinsonb5, 2020-09-24:
OK, yes, that makes sense.
First, though, we'd better check that my diagnosis is correct.

@JorsuSHerluN could you try this version, and let me know if it works with your monitor, please?
[minimig_ntsc_clk.zip](https://github.com/mist-devel/mist-binaries/files/5279243/minimig_ntsc_clk.zip)

JorsuSHerluN, 2020-09-25:
> OK, yes, that makes sense.
> First, though, we'd better check that my diagnosis is correct.
> 
> @JorsuSHerluN could you try this version, and let me know if it works with your monitor, please?
> [minimig_ntsc_clk.zip](https://github.com/mist-devel/mist-binaries/files/5279243/minimig_ntsc_clk.zip)

This core works well on my monitor. 
Thank you very much.

closed 2020-09-26 by JorsuSHerluN